thomluther/ha-anker-solix #242: Schedule does not work - displays no schedule available

Flirina, 2025-04-05:
### System Health details

## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1690
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 62.3 GB
disk_used | 55.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | RTSP Simple Server Add-on (v0.17.6), Terminal & SSH (9.16.0), File editor (5.8.0), Duck DNS (1.18.0), Mosquitto broker (6.5.0), eufy-security-ws (1.9.1), Advanced SSH & Web Terminal (20.0.2), InfluxDB (5.0.2), Grafana (10.5.0), Solix 2 MQTT (1.8.2), MQTT Explorer (browser-1.0.3), Govee to MQTT Bridge (2025.01.04-2c39a50f), SQLite Web (4.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 14
views | 11
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp1s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), vethf215dc1 (disabled), vethe5cb81c (disabled), veth85bdc6b (disabled), veth8404e45 (disabled), vethb0b64be (disabled), veth122bb03 (disabled), veth369eda7 (disabled), vethbcc5838 (disabled), veth4740919 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp1s0 (192.168.3.2/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), vethf215dc1 (), vethe5cb81c (), veth85bdc6b (), veth8404e45 (), vethb0b64be (), veth122bb03 (), veth369eda7 (), vethbcc5838 (), veth4740919 ()
ipv6_addresses | lo (::1/128), enp1s0 (fe80::f9c8:2567:8f8f:17a7/64), docker0 (fe80::42:1eff:fe02:e522/64), hassio (fe80::42:1fff:fe86:6edf/64), vethf215dc1 (fe80::e4ea:b4ff:fe2a:95ce/64), vethe5cb81c (fe80::ac12:66ff:fe92:a05/64), veth85bdc6b (fe80::989e:99ff:feb9:9730/64), veth8404e45 (fe80::a0d2:34ff:fe49:c490/64), vethb0b64be (fe80::f403:d1ff:fe74:783f/64), veth122bb03 (fe80::70dc:1ff:fe34:bc2c/64), veth369eda7 (fe80::ecf2:5cff:fe0c:6490/64), vethbcc5838 (fe80::7486:6aff:fed0:acc/64), veth4740919 (fe80::b431:faff:fe1b:f87b/64)
announce_addresses | 192.168.3.2, fe80::f9c8:2567:8f8f:17a7

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24. März 2025 um 14:50
-- | --
current_recorder_run | 4. April 2025 um 21:43
estimated_db_size | 4055.93 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --

</details>

### Integration version

2.5.7

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hi,
since some time I am unable to  display the correct schedule in home assistant. It always says no schedule available. I already tried to delete and recreate the schedule completely, but it still says no schedule available.

Debug logs attached.
Setup:
2x E1600 Solarbanks with HM 1500 (faked as the anker inverter for a dual setup).

### Reproduction steps

1. create schedule in anker app
2.no schedule is displayed in home assistant
3. also no updates when schedule is modified or deleted and recreated

[home-assistant_anker_solix_2025-04-05T06-11-32.090Z.log](https://github.com/user-attachments/files/19614028/home-assistant_anker_solix_2025-04-05T06-11-32.090Z.log)

...

### Debug logs

```text
attached as file
```

thomluther, 2025-04-05:
Looking at the last log entries in debug mode, the schedule is available and also changes are applied without error.
I don't know why you no longer see it in your dashboard. It is part of the entity attribute, you can check that also in the developer tabs.

But you have running some 'dirty' automatiion which is kicking in endlessly while previous is still active:
```
2025-04-05 06:30:10.788 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:16.780 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:22.775 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:28.785 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:34.779 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:40.789 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:52.778 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:30:58.780 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:31:04.780 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:31:10.782 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:31:16.775 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
2025-04-05 06:31:22.786 WARNING (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Already running
```

I think that is not really well from a performance and event handling perspective.

Furthermore you had many api errors at the beginning where I have no idea why they occur, it was without debug. But it is also triggered from your automation:
```
custom_components.anker_solix.solixapi.errors.AnkerSolixError: (10004) Anker Api Error: Failed to request.
2025-04-05 00:05:10.971 ERROR (MainThread) [homeassistant.components.automation.beta_6_6] BETA 6.6: Error executing script. Unexpected error for choose at pos 1: (10004) Anker Api Error: Failed to request.
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 524, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 715, in _async_step_choose
    await self._async_run_script(script)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 632, in _async_run_script
    result = await self._async_run_long_action(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 624, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1828, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 460, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    self._handle_exception(
    ~~~~~~~~~~~~~~~~~~~~~~^
        ex, continue_on_error, self._log_exceptions or log_exceptions
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 556, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 524, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 715, in _async_step_choose
    await self._async_run_script(script)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 632, in _async_run_script
    result = await self._async_run_long_action(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 624, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1828, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 460, in async_run
[... 36 lines cut ...]
             ^^^^^^^^^^
  File "/config/custom_components/anker_solix/switch.py", line 326, in async_turn_on
    resp = await self.coordinator.client.api.set_home_load(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<11 lines>...
    )
    ^
  File "/config/custom_components/anker_solix/solixapi/schedule.py", line 1154, in set_home_load
    return await self.set_device_parm(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/config/custom_components/anker_solix/solixapi/schedule.py", line 251, in set_device_parm
    await self.apisession.request(
        "post", API_ENDPOINTS["set_device_parm"], json=data
    )
  File "/config/custom_components/anker_solix/solixapi/session.py", line 474, in request
    errors.raise_error(data)
    ~~~~~~~~~~~~~~~~~~^^^^^^
  File "/config/custom_components/anker_solix/solixapi/errors.py", line 114, in raise_error
    raise error(f"({code}) {prefix}: {data.get('msg', 'Error msg not found')}")
custom_components.anker_solix.solixapi.errors.AnkerSolixError: (10004) Anker Api Error: Failed to request.

```

Flirina, 2025-04-05:
I think I found the error.
when I change the first line to:

{% set entity = 'sensor.solarbank_e1600_ost_einspeisevorgabe' %}

it works and I can see the complete schedule.
it didn't work with the systems entity: 

 {% set entity = 'sensor.system_flirina_sb_einspeiseleistung' %}

which was working before. I was assuming to query the schedule from the system and not from each of the individual solar bank "einspeisevorgaben"....

regarding the errors..thanks, but nevermind, its the alkly automation which was "verpfuscht" a little by me.. but its working like a charm :)

thomluther, 2025-04-06:
> regarding the errors..thanks, but nevermind, its the alkly automation which was "verpfuscht" a little by me.. but its working like a charm :)

I would try to get rid of the warnings anyway. They spam your log...

> I was assuming to query the schedule from the system and not from each of the individual solar bank "einspeisevorgaben"....

The schedule was always only in the attribute of the individual device entities for System Einspeisevorgabe, never in any system entity. Not sure why it was working before for you, maybe you mix up the entities names. You can check in the developer tab which entities have which attributes.

If there are no further problems, pls close this issue again.
Thx

closed 2025-04-06 by Flirina