thomluther/ha-anker-solix #243: bug: Feed in broken after i manually adjusted the amount

EinfachHans, 2025-04-05:
### System Health details

## System Information

version | core-2025.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1626
Downloaded Repositories | 13

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.4
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 938.2 GB
disk_used | 9.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2), File editor (5.8.0), ArgonOne V3 Active Linear Cooling (30a), Cloudflared (5.2.12), Get HACS (1.3.1), OpenThread Border Router (2.13.0), Matter Server (7.0.0), Mosquitto broker (6.5.0), Zigbee2MQTT (2.2.1-1), Zigbee2MQTT (2.2.1-1), SQLite Web (4.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
[... 128 lines cut ...]
2025-04-05 19:46:30.761 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'home_info': {'home_name': 'Home', 'home_img': '', 'charging_power': '0.00', 'power_unit': 'W'}, 'solar_list': [], 'pps_info': {'pps_list': [], 'total_charging_power': '0.00', 'power_unit': 'W', 'total_battery_power': '0.00', 'updated_time': '', 'pps_status': 0}, 'statistics': [{'type': '1', 'total': '5.26', 'unit': 'kwh'}, {'type': '2', 'total': '5.25', 'unit': 'kg'}, {'type': '3', 'total': '2.05', 'unit': '€'}], 'topology_type': '1', 'solarbank_info': {'solarbank_list': [{'device_pn': 'A17C0', 'device_sn': 'AZV6Y60E23100125', 'device_name': 'Solarbank E1600', 'device_img': 'https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png', 'battery_power': '100', 'bind_site_status': '', 'charging_power': '0', 'power_unit': 'W', 'charging_status': '1', 'status': '1', 'wireless_type': '1', 'main_version': '', 'photovoltaic_power': '0', 'output_power': '0', 'create_time': 1743610811, 'set_load_power': '', 'sub_package_num': 0, 'output_cutoff_data': 10, 'is_display': True, 'bat_charge_power': '0'}], 'total_charging_power': '0', 'power_unit': 'W', 'charging_status': '0', 'total_battery_power': '1.00', 'updated_time': '2025-04-05 19:45:46', 'total_photovoltaic_power': '0', 'total_output_power': '0.00', 'display_set_power': False, 'battery_discharge_power': '0', 'ac_power': '0', 'to_home_load': '0', 'is_display_data': True, 'solar_power_1': '0', 'solar_power_2': '0', 'solar_power_3': '0', 'solar_power_4': '0', 'other_input_power': '0', 'micro_inverter_power': '0', 'micro_inverter_power_limit': '0', 'micro_inverter_low_power_limit': '0', 'grid_to_battery_power': '0', 'pei_heating_power': '0', 'backup_info': {'start_time': 0, 'end_time': 0, 'full_time': 0}}, 'retain_load': '0W', 'scene_mode': 0, 'home_load_power': '0', 'updated_time': '01-01-0001 00:00:00', 'power_site_type': 2, 'site_id': '7c7b85dc-0463-4294-b09c-3ea8ba2b74a4', 'powerpanel_list': [], 'grid_info': None, 'is_downgrade': False, 'error_code': 0, 'smart_plug_info': None, 'feature_switch': None, 'other_loads_power': '0', 'priority_discharge_switch': 0, 'display_priority_discharge_tips': 0, 'priority_discharge_upgrade_devices': '', 'style_id': 0, 'is_show_priority_discharge': 1}, 'trace_id': 'ad8a3ddd9aeb8a8b14a87ff1c84c740a'}
2025-04-05 19:46:30.762 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator han*** request statistics: 182 last hour, 6 last minute
2025-04-05 19:46:30.762 DEBUG (MainThread) [custom_components.anker_solix] Coordinator han*** data: {'7c7b85dc-0463-4294-b09c-3ea8ba2b74a4': {'type': 'system', 'site_info': {'site_id': '7c7b85dc-0463-4294-b09c-3ea8ba2b74a4', 'site_name': 'Balkonkraftwerk', 'site_img': '', 'device_type_list': [3], 'ms_type': 1, 'power_site_type': 2, 'is_allow_delete': True, 'support_device_models': ['A17C0', 'A5143', 'A1771'], 'current_site_device_models': ['A17C0']}, 'site_type': 'solarbank', 'site_admin': True, 'data_valid': True, 'requeries': 0, 'energy_offset_seconds': -183, 'energy_offset_check': '2025-04-05 19:46:30', 'energy_offset_tz': 0, 'home_info': {'home_name': 'Home', 'home_img': '', 'charging_power': '0.00', 'power_unit': 'W'}, 'solar_list': [], 'pps_info': {'pps_list': [], 'total_charging_power': '0.00', 'power_unit': 'W', 'total_battery_power': '0.00', 'updated_time': '', 'pps_status': 0}, 'statistics': [{'type': '1', 'total': '5.26', 'unit': 'kwh'}, {'type': '2', 'total': '5.25', 'unit': 'kg'}, {'type': '3', 'total': '2.05', 'unit': '€'}], 'topology_type': '1', 'solarbank_info': {'solarbank_list': [{'device_pn': 'A17C0', 'device_sn': 'AZV6Y60E23100125', 'device_img': 'https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png', 'battery_power': '100', 'bind_site_status': '', 'charging_power': '0', 'power_unit': 'W', 'charging_status': '1', 'status': '1', 'wireless_type': '1', 'main_version': '', 'photovoltaic_power': '0', 'output_power': '0', 'create_time': 1743610811, 'set_load_power': '', 'sub_package_num': 0, 'output_cutoff_data': 10, 'is_display': True, 'bat_charge_power': '0', 'alias_name': 'Solarbank E1600', 'parallel_home_load': '0', 'current_home_load': '0'}], 'total_charging_power': '0', 'power_unit': 'W', 'charging_status': '0', 'total_battery_power': '1.00', 'updated_time': '2025-04-05 19:45:46', 'total_photovoltaic_power': '0', 'total_output_power': '0.00', 'display_set_power': False, 'battery_discharge_power': '0', 'ac_power': '0', 'to_home_load': '0', 'is_display_data': True, 'solar_power_1': '0', 'solar_power_2': '0', 'solar_power_3': '0', 'solar_power_4': '0', 'other_input_power': '0', 'micro_inverter_power': '0', 'micro_inverter_power_limit': '0', 'micro_inverter_low_power_limit': '0', 'grid_to_battery_power': '0', 'pei_heating_power': '0', 'backup_info': {'start_time': 0, 'end_time': 0, 'full_time': 0}}, 'retain_load': '0W', 'scene_mode': 0, 'home_load_power': '0', 'updated_time': '01-01-0001 00:00:00', 'power_site_type': 2, 'site_id': '7c7b85dc-0463-4294-b09c-3ea8ba2b74a4', 'powerpanel_list': [], 'grid_info': None, 'is_downgrade': False, 'error_code': 0, 'smart_plug_info': None, 'feature_switch': None, 'other_loads_power': '0', 'priority_discharge_switch': 0, 'display_priority_discharge_tips': 0, 'priority_discharge_upgrade_devices': '', 'style_id': 0, 'is_show_priority_discharge': 1, 'site_details': {'price': 0.39, 'site_co2': 0, 'site_price_unit': '€', 'price_type': 'fixed', 'current_mode': 0}}, 'AZV6Y60E23100125': {'device_sn': 'AZV6Y60E23100125', 'type': 'solarbank', 'site_id': '7c7b85dc-0463-4294-b09c-3ea8ba2b74a4', 'is_admin': True, 'device_pn': 'A17C0', 'generation': 1, 'battery_soc': '100', 'battery_capacity': '1600', 'battery_energy': '1600', 'charging_power': '0', 'power_unit': 'W', 'charging_status': '1', 'charging_status_desc': 'bypass', 'status': '1', 'status_desc': 'online', 'wireless_type': '1', 'input_power': '0', 'output_power': '0', 'output_cutoff_data': 10, 'bat_charge_power': '0', 'alias': 'Solarbank E1600', 'name': 'Solarbank E1600', 'set_output_power': '0', 'set_system_output_power': '0', 'data_valid': True, 'solarbank_count': 1, 'bt_ble_mac': 'F49D8A11B596', 'wifi_online': True, 'charge': False, 'bws_surplus': '0', 'sw_version': 'v2.0.9', 'tag': '', 'rssi': '', 'auto_upgrade': False, 'is_ota_update': False, 'ota_version': 'v2.0.9', 'ota_children': [{'device_type': 'A17C0_mcu', 'need_update': False, 'force_upgrade': False, 'rom_version_name': 'v2.0.9'}, {'device_type': 'A17C0_esp32', 'need_update': False, 'force_upgrade': False, 'rom_version_name': 'v0.0.9.8'}], 'wifi_name': 'Einfach Wlan', 'wifi_signal': '58', 'power_cutoff_data': [{'id': 1, 'is_selected': 1, 'output_cutoff_data': 10, 'lowpower_inpu [... cut ...] web.de', 'nickname': 'han***', 'country': 'DE', 'server': 'https://ankerpower-api-eu.anker.com', 'currency_list': [{'symbol': '$', 'name': 'USD'}, {'symbol': '€', 'name': 'EUR'}, {'symbol': 'zł', 'name': 'PLN'}], 'default_currency': {'symbol': '€', 'name': 'EUR'}, 'requests_last_min': 6, 'requests_last_hour': 182, 'use_files': False, 'has_unread_msg': False, 'system_msg': False, 'device_msg': False}}
```

### Diagnostics dump

I don't have the option to download diagnostics data for the device 🤔

EinfachHans, 2025-04-06:
Hey @thomluther, thanks for the response - Okay i will do it. How does the schedule work with custom set values? For example when i have a slot for 12:00 - 14:00 Uhr for 500W and i change it on 11:59 via the integration to 400W, will it then be changed a minute later because of the schedule?

thomluther, 2025-04-06:
The active slot according to the HA instance time will be changed. It doesn't make sense to have some seconds tolerance since the time Offset between HA and Solarbank is unknown.

thomluther, 2025-04-06:
> [@thomluther](https://github.com/thomluther) it was more of a theoretical question if the general schedule will override the manually set value (Einspeisevorgabe) at some point

I don't understand the question. The manually set value is part of the overall schedule. There is no way to set an output without modifying the general schedule. If there is no schedule slot, the output will always be 200 W

EinfachHans, 2025-04-06:
What do you mean with

> There is no way to set an output without modifying the general schedule.

When i'm currently in a slot 10:00-15:00 with 300W and i manually set the "Einspeisevorgabe" to 500W, would this be valid?

thomluther, 2025-04-07:
> What do you mean with
> 
> > There is no way to set an output without modifying the general schedule.
> 
> When i'm currently in a slot 10:00-15:00 with 300W and i manually set the "Einspeisevorgabe" to 500W, would this be valid?

Yes, this is valid, it will change the slot to 500W

thomluther, 2025-04-07:
I also heard, that with 2 Solarbank Systems the latest firmware changed the max setting down to 800 W which was 1600 W before.
So if you preset over 800 W, I think it will either be ignored, or limit to 800 W no matter what you specify.
The same is valid below the minimum of 100/150W. You can set it below your specific minimum, but the limits are enforced by the firmware and the configured inverter type.

EinfachHans, 2025-04-07:
> Yes, this is valid, it will change the slot to 500W

Ahh, this setting will **update** the current slot? So when i set it the 10:00-15:0 slot will be updated to 500W and will also be 500W on the next day when not changed again?

thomluther, 2025-04-07:
Correct, therefore I say there is no way to change the output setting directly. It only works via the active slot in the schedule and any change you will apply to the slot is permanent for that slot until you change it again.
In fact, any change in the control entities that are related to the charge/discharge behavior are attributes of the schedule. So changing them will change the active slot of your schedule saved to the cloud and the device.

EinfachHans, 2025-04-07:
Got it, thank you for explaining! 😊

closed 2025-04-07 by EinfachHans