thomluther/ha-anker-solix #248: Add initial support for Solarbank 3

thomluther, 2025-04-10:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Solarbank 3 was announced in April 8 2025 and is expected in May.
It comes with a couple of new features (Smart Mode), dynamic tariff support, AC charging etc.

### Describe the solution you'd like

Initially support the new system for known features.

### Describe alternatives you've considered

None

### Additional context

N/A

thomluther, 2025-04-10:
System owners will have to provide system exports from their system at various situations to gather sufficient information for proper adjustments of the library.
Cloud updates as well as App updates still need to be done for the Solarbank 3 support.

charlyschulte, 2025-04-14:
hey @thomluther maybe i can provide thore, because i already have the solix 3 here for testing purposes

thomluther, 2025-04-14:
You can try, but I don't think that will work because the test systems typically run with a beta App as well as with a special cloud server.
The cloud server is hard coded, you would have to change that in the apitypes.py and see if it works. I have no idea, if the public key will work for the beta cloud server and the country assignment may also be wrong (so the beta cloud server must be used for all countries...)

thomluther, 2025-04-27:
> Maybe the differences are not too big as the AC charging had already beein intruduced with the E1600AC and the real new thing is the AI stuff, so functional maybe we're lucky and integration will work quickly?

Maybe, if the data publish mechanisms do not change like with the SB2 a lot of commonality with SB2 should work out of the box.
Definitely I will need system exports as owner account from various conditions and especially how the new smart mode plan will look like or dynamic tariffs stuff.
You may have to try various Api calls directly to see what is working and giving new data points.
So nothing of the new stuff will work quickly and depends alot from community contributions, Api exploration and testing.

macmaniacs, 2025-04-28:
> You may have to try various Api calls directly to see what is working and giving new data points.
So nothing of the new stuff will work quickly and depends alot from community contributions, Api exploration and testing.

Well, happy to help, just need to be told exactly how to do it, noob here.

idl0r, 2025-05-17:
Let me know what you guys need and I can try to help and provide some data. I just got mine and have it running but only in AC mode. No panels connected or anything else, really AC only.

charlyschulte, 2025-05-17:
I am also willing to help, now I am on the production server. The problem is that I am not really experienced in python and setting up the project hasn't worked for me first try

thomluther, 2025-05-17:
Is the SB3 already supporrted in official App version 3.7.0? Nothing is listed there...
You can start with a [Export systems action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action) once you have a system with SB3 on the normal Anker App (standard cloud servers). The zip file will contain data for the known endpoints.

You can also explore the Anker Solix cloud Api directly within your HA instance via the integration's [Api request action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#api-request-action).

I assume the new AI mode requires additional endpoints to query and control settings. You will have to explore those.
If someone can run the App on a rooted device or emulator with trusted TLS certificates installed, you can sniff the http traffic with an App like PCAPdroid. [Here ](https://github.com/krikk/hoymiles-ms-a2-to-mqtt?tab=readme-ov-file#how-to-sniff-the-app-traffic)is some information how this could be done.
PCAPdroid will reveal the endpoints depending on which actions you do in the app. But the payload is encrypted unfortunately, so parameters must be explored manually...
For iphones someone used used "Surge" (Can get Flutter-App Traffic),

charlyschulte, 2025-05-17:
Yes, the Powestation is Not listed but still supported by 3.7.0
I will try that and give Feedback 
Thanks for the detailed description

thomluther, 2025-05-20:
Hi
On my production server which is on Frontend 20250516.0, there seems to be an issue with Action responses in the UI. They are not shown anymore. I opened a [frontend issue](https://github.com/home-assistant/frontend/issues/25542) for that.
If you experience the same, please update that issue.

If you have the same issue, the problem is that you cannot see when the System Export is finished. This can take 1-5 Minutes, depending on how much request throttling must be applied for the energy queries.

If you just wait long enough and click on the button Copy to Clipboard (as template), you can paste the response into an editor to see it. It will show you URL path that you can extend to your HA Url to download the zip file.

thomluther, 2025-05-23:
Is there someone who can provide a SB2 export zip?
The frontend issue with the Actions panel described above has been fixed. It was related to auto_entities card. In case you have that installed, make sure to update latest repository information and install the latest fix from today to resolve the response display issue in the Actions tab of HA developer tools.

charlyschulte, 2025-05-23:
here you go
i hope that helps

[spaxxx_2025-05-23_1341.zip](https://github.com/user-attachments/files/20409739/spaxxx_2025-05-23_1341.zip)

thomluther, 2025-05-23:
Here are some new fields in scen_info response that need explanation:
```json
    "solarbank_pps_info": {
      "pps_list": null,
      "total_charging_power": "",
      "total_output_power": "",
      "total_battery_power": "",
      "total_pv_input_power": "",
      "total_home_load_power": "",
      "total_grid_to_battery": ""
    },
    "user_scene_mode": 7,
    "is_dynamic_price_down_grade": false,
    "aiems_profit": {
      "unit": "\u20ac",
      "aiems_profit_total": "0.76",
      "percentage": "322.22",
      "show_percentage": true,
      "aiems_self_use_diff": "0.58"
    },
``` 
I wonder if Anker plans to support power stations attached to SB3?
User mode 7 is AI mode? In that case there is missing translation for mode 6 which I have not seen yet...Only modes 1-5 are used for SB2...

Furthermore, have you checked the App with the member account? Does that display the AC Charge from grid correctly?
Your export is showing data that indicates it is charing from grid, but required fields are still "0" like for the SB2 AC. This is a bug to my opinion, and may be present only for shared accounts (App NOT used by owner account, which may trigger updates of such fields...)
If the AC flow into battery disappears on the home screen when only using the member account, you would have something to address this issue to Anker because it exists since SB2 AC but this field does not seem to be used on Home Screen so far. The AC never showed any flow into the battery from Grid...

Everyone: Please open issues for bug fixing or feature requests if you use the SB3 in the integration. I don't know what is working and what still has to be done to support SB3 correctly...

thomluther, 2025-05-25:
If you create a system export, it will be mor useful to do it as owner account to get more data.
If user mode 7 is the Smart Mode, user mode 6 could be the new Time Slot mode for dynamic tariffs. Both require a Smartmeter. I will need example exports when each of them is configured with all settings to see if they are configured with schedule object enhancements.

kitokirisaki, 2025-05-25:
I would like to help, but currently I don't even get correct data from the box. The SB Datenzeit updates but nothing else.

thomluther, 2025-05-26:
OK thx, those are then pretty much self explainable and information only.
The new Time Slot mode (when active) and where all its settings are listed will be required. If they are not part of the common schedule object, it is unclear how to obtain them and change them eventually.
The same is valid with active smart mode. Even if that will be part of the common schedule object, it is not clear if that can be toggled/enabled via the Api, since it requires some initial permissions in the app from what I have seen, like location data etc....
I also read it requires Google Maps on Android phones to be usable, but not sure about that...

kitokirisaki, 2025-05-26:
> I have no idea how you can fix it if you don't get data from your account. As long as there are responses, the account exists in the cloud. So re-adding it to the integration probably does not make any difference.

Ok, so I checked a bit more on what data is there and what not.
Look's like the Solarbank Site does get updates, but the Smart Meter and the Solarbank 3 E2700 Pro do not get any updates.
If I reload the integration it does updates but then the same issue comes up.

kitokirisaki, 2025-05-26:
> If you create a system export, it will be mor useful to do it as owner account to get more data.
If user mode 7 is the Smart Mode, user mode 6 could be the new Time Slot mode for dynamic tariffs. Both require a Smartmeter. I will need example exports when each of them is configured with all settings to see if they are configured with schedule object enhancements.

I tried the Schedule Mode with the dynamic tariffs, but the status is 8 not 6.

idl0r, 2025-05-27:
> Is the SB3 already supporrted in official App version 3.7.0? Nothing is listed there... You can start with a [Export systems action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action) once you have a system with SB3 on the normal Anker App (standard cloud servers). The zip file will contain data for the known endpoints.
> 
> You can also explore the Anker Solix cloud Api directly within your HA instance via the integration's [Api request action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#api-request-action).
> 
> I assume the new AI mode requires additional endpoints to query and control settings. You will have to explore those. If someone can run the App on a rooted device or emulator with trusted TLS certificates installed, you can sniff the http traffic with an App like PCAPdroid. [Here ](https://github.com/krikk/hoymiles-ms-a2-to-mqtt?tab=readme-ov-file#how-to-sniff-the-app-traffic)is some information how this could be done. PCAPdroid will reveal the endpoints depending on which actions you do in the app. But the payload is encrypted unfortunately, so parameters must be explored manually... For iphones [someone used "Surge"](https://github.com/thomluther/anker-solix-api/issues/18#issue-2139505404) (Can get Flutter-App Traffic), that may work without rooted device...

systems export of SB3:
https://nc.qasl.de/index.php/s/d4GYaDCEn7qTDYg

I hope it helps

thomluther, 2025-05-30:
> Is it possible to seperat additional accus?

There is no Api query to report them separately. Only the count is reported, but no type, SN, temp or other device info you may see in the app.

Jeepersfinest, 2025-05-30:
mhhh....i can see this in my app: Geräte -> SB3 -> Akku : see picture

[image]

thomluther, 2025-05-30:
For those who have the 2.8.0 version running with admin account:
Please try with SB3 whether the known SB2 modes can be switched, and whether you can also modify the known Price Types and Usage Time plan, so that it really also applied to SB3 correctly.
If that does not work anymore with SB3, it would mean the approach for changing the known settings through the Api also have changed for SB3

frankymusik, 2025-05-30:
For me, it looks like the integration of the Solix 3 Pro is already working well. Of course, I can't check every detail...

idl0r, 2025-05-30:
At least the "Einspeisevorgabe", I guess that's power output, seems wrong on the SB3. It shows 200W. Default is 800W and I am running 1200W.
Also adding the SB3 to the energy dashboard seems to not work properly as HA complains about not being reset, wrong class etc.

thomluther, 2025-05-31:
> Yeah but that's like the maximum power that can go out, right? So that's set to 800W by default but I set it up to 1200W max. This value however is at 200 all the time and that looks wrong to me. If it is, what I think it is, then it actually is wrong.

Actually the set inverter limit = max. AC output to the house, is not provided reliably by the Api, and for sure it cannot be set. At least I'm not aware of any endpoint that can query the limit and change it, But that does not mean it does not exist....You may explore if there are andpoints to query and set the AC Limit, then I can integrate it (not only for SB3).

The output_preset = Einspeisevorgabe is what should be feed into the house when manual plan or smart plug plan is used...
This entity is unavailable if other plans are active. The frontend may display the default setting, also the Api may report system output preset as default, but actually it is not applied in those plans, only if the smart meter fails.
So that is all working as it should.

thomluther, 2025-06-02:
I don't get behind the new time_slot mode. Even if it is enabled (mode 8), the Api does not show any plan in the schedule object:
```json
    "time_slot": null,
    "schedule_mode": {
      "last_select_mode": 8
    },
```
So I have no idea where to find the settings of the time_slot mode, nor how they can be modified via Api. Another inconsistent implementation of the Api.
Furthermore I do not find any information about the actual dynamic price as displayed in the Api.
So at this point, Switching to the new modes may be possible (if they have been configured previous in the App), but that's it.
It may also fail, since it cannot be validated if a time_slot plan has been defined...
Still Open: What is usage mode 6? Haven't found any explanation yet for it, but must be unique to SB3 if used at all.
At this point, I don't think all those new features can be supported through the Api.
It also still shows the known problem from AC for some Api fields.
E.g. the Grid to battery power is only provided in the response if the App is open with the owner account, otherwise it is always 0 even if the battery is still AC charging...
This Api is just crap...

thomluther, 2025-06-04:
@DenisMtfl Thx for the export. I see you have no smartmeter, only plugs.
Couple of questions:
1. Which Usage modes are usable in the App if only plugs are in the system? Can you enable Time Slot mode, Usage Time mode, Emergency backup option, Smart mode?
2. Which Price Types are usable? Can you enable/configure dynamic prices / provider if no smart meter in the system?
3. Your device_parm_13 data looks interesting. Any idea what ai_ems status 3 could mean? If you cannot use smart mode, it maybe means that learning never started?
```json
  "data": {
    "param_data": {
      "step": 10,
      "ai_ems": {
        "status": 3
      },
      "max_load": 1200,
      "min_load": 0,
      "data_auth": false,
      "mode_type": 7,
      "blend_plan": [
        {
          "week": [0,1,2,3,4,5,6],
          "index": 0,
          "ranges": [
            {
              "power": 0,
              "end_time":"05:00",
              "start_time":"00:00"
            },
            {
              "power": 60,
              "end_time":"21:00",
              "start_time":"05:00"
            },
            {
              "power": 150,
              "end_time":"24:00",
              "start_time":"21:00"
            }
          ]
        }
      ],
      "reserved_soc": 0,
      "custom_rate_plan": [
        {
          "week": [0,1,2,3,4,5,6],
          "index": 0,
          "ranges": [
            {
              "power": 240,
              "end_time":"24:00",
              "start_time":"00:00"
            }
          ]
        }
      ],
      "default_home_load": 200
    }
  },
```
If I compare the scene info data from a smart mode enabled system, it shows following:
```json
      "ai_ems": {
        "enable": true,
        "status": 5
      },
```
So if 3 means 'Not initialiazed or Untrained', 5 could stand for Training completed. Eventually there is 4 or another value while training is running? Someone that is in this situation would need to trigger an export or check what the Scene Info query data is showing at that time to validate the training state. Then it remains open what 1 and 2 could mean if they exist?

6. What could mode type 7 mean in your output? Scene mode 7 typically stands for Smart mode, but your active scene mode is 3 = Manual as given in the scene_info. So why is this output showing 7, while the parm_6 output is showing `"mode_type": 3`. This does not really fit together, more Api inconsistency???
7. Can you confirm your Solarbank is set to 1200 W as shown in the `"max_load": 1200`? I wonder if the schedule object can also be used to toggle the max load setting? I only tried that with SB1 devices and that setting was simply ignored. Eventually it works for SB2 and 3 when providing allowed Limits? That would provide a way to toggle the output limit via Api if accepted by the device...

thomluther, 2025-06-04:
> In the official Anker app, I can configure time slots, smart mode, and other settings.
> No price type is defined in my case.

I'm interested in what is allowed by the App in your setup without smart meter, because I would need to block unsupported modes or price types if no smart meter is installed.
I read that Smart mode is not possible without smart meter. Could you enable it and start the learning phase also with just smart plugs installed?
If you can, it will be helpful to create another export while it is in learning phase for 24h to see which status the ai_ems is reporting...

I think dynamic price providers should not be dependent on the smart meter. So could you configure them, even if just used for saving calculations? I think Usage Time or Time Slot mode cannot be enabled in your system, because they support AC charge, which typically must be able to recognize surplus. But time slot mode could also work without smart meter, since AC charge is only determined by cheap price slots after you configured dynamic prices or Usage Time tariffs.

So it is important to understand, which usage modes and price types must be blocked if no smart meter installed?

rottweilerken, 2025-06-06:
Hi,
I know it's not the place here, but I want to thank you very very much, for your time and effort, making this plugin available for HA!
I also recently installed 2 x SB3 with an additional battery. Solar panels are for next month, but already dynamic pricing contract.
After talking to support about enabling the possibility to add the Shelly EM3Pro in Belgium, I could add it now. But as you cannot share the shelly between 2 SB's, it's pretty inefficient to manage it through the Anker App. Thank you therefor as I can enable charge and discharge power through your plugin!
FYI, without a Smart Meter or Shelly Meter, the very single and only option possible in the Anker app is: manual.
Not even the possibility to have it charge through the Anker App if you don't attach a smart meter.

thomluther, 2025-06-06:
> Not even the possibility to have it charge through the Anker App if you don't attach a smart meter.

Glad that it helps you to customize your solarbank. So you are saying even the Emergency backup option is not available in the app without smart meter? That is weird. I would open that as ticket if this is true. There is no need for a smart meter to fully charge from AC as quickly as possible for the configured time slot...

Marianp90, 2025-06-15:
>  But as you cannot share the shelly between 2 SB's (yet), it's pretty inefficient to manage it through the Anker App. Thank you therefor as I can enable charge and discharge power through your plugin in HA, making it smart, exactly as I want FYI, without a Smart Meter or Shelly Meter, the very single and only option possible in the Anker app is: manual. Not even the possibility to have it charge through the Anker App if you don't attach a smart meter.

This could be a solution, you can simulate a Shelly with Shelly as input device. Hence, you may have a second Shelly to connect to your second Solarbank:
https://github.com/sdeigm/uni-meter

thomluther, 2025-06-15:
Not sure if uni-meter works for Solix since Data is needed from Shelly cloud which cannot be emulated.
Also Powerfox via cloud API is too slow, can take 30 sek. to get updated data. The Integration allows only 60 sek interval at least as far as I know.

Marianp90, 2025-06-15:
I haven't tried yet, whether data is pulled from Shelly Cloud or locally. I hope that the Cloud account is only used for authentication, but data flow would be local - let me try the next day's to implement that.
Powerfox in my case is used locally with one update per second (haven't upgrade to their latest firmware which means I use the device only locally not with their app anymore - it's from a time where I haven't started with HA 😉).

Marianp90, 2025-06-15:
@DenisMtfl at least worth a try? Please keep us posted. I will try mid of next week.

stachel15, 2025-06-18:
https://support.ankersolix.com/de/s/article/Welche-Smart-Meter-sind-mit-der-Solarbank-3-E2700-Pro-kompatibel

The Solarbank 3 E2700 Pro is currently compatible with the Anker Smart Meter, Shelly 3EM and Shelly Pro 3EM. Other smart meters cannot currently be connected to the Anker app.

rottweilerken, 2025-06-18:
Would it be possible to have these settings available and changeable? I need it to limit the output of the SB3 when the battery is fully charged but the panels are still producing.. in this case, with dynamic pricing and nobody at home it sends energy back to the grid and I have to pay for that.. so I'd like to limit it with this feature.. works through the Anker mobile app..
Let me know if I can help testing API calls..

[image]
[image]

rottweilerken, 2025-06-19:
> However, what you want to achieve is possible by switching to the manual usage mode with any output you specify, and that will be used as max output as far as I know. 

This doesn't work when no smart meter is attached, AND the battery is at 100% AND the PV production is above the consumption..
Putting manual with 0W or 10W, always outputs 800W.. 
I've contacted the support on that as it seems a bug, also on 1.0.0.18, waiting their reply..
When in Smart Mode (self consumption with dynamic pricing in the app) with a smart meter attached, this work well, the system reduces the PV production, or shuts down PV ports to reduce it, and the output is exactly what you consume..

thomluther, 2025-06-19:
If you find a way to modify the device max load to one of the defined options, let me know ans I can implement it. But you should keep in mind, that change of this limit probably goes to device EEprom and change cycles may be limited. Modify it 1-2 times a day is maybe not a problem over lifetime, but I would not stress it too much.
Set device parm changes became tricky since SB2 AC.
Before you could modify only desired values and return the whole schedule object. Since new plans of AC model, whenever you return use_time or backup fields, the other plans are ignored. So depending on which plan you modify, you have to construct the schedule object accordingly. That may be even more difficult with new SB3 plans.
I know that top level fields can eventually always be changed?
Also the dynamic price and aiems fields seem to be info or read only. The dynamic price provider is changed with the price update endpoint, but only possible when dynamic price type is being activated at same time. No way to clear the provider settings again with that endpoint. But I believe provider clearing is not possible in the App either.
The time_slot plan doesn't seem to work at all, I never saw an object there, even if time slot made was active in the App. FW or cloud bug???
I also need a translation for mode_type 6 whenever this is shown in the schedule parts or the scene_mode field in scen_info response.

thomluther, 2025-06-26:
> > After verifying the issue you described, we can confirm that in Custom Mode, once the battery is fully charged, the system will automatically switch to bypass mode. This means that, in this case, regardless of the wattage settings you enter in Custom Mode, the unit will bypass the custom settings and continue outputting at its maximum capacity. For your Solarbank 3 Pro, this output is approximately 800W, which explains why you are consistently seeing 800W instead of your custom preset.
> 
> Is anybody else seeing this behavior ?

That is a normal behavior to my knowledge since all generations.
There is only the new device option to avoid export to the grid. Then PV should be throttled to whatever is the home demand if the battery is full or limited in charging.
If that new device mode does not work in custom mode, or if no smart meter is installed, then this is a defect that you need to address to Anker. If the new device option is set, it should be used in ANY mode since each mode has either an automatic or custom home demand value, that should not be exceeded if that option is set.

chrisvw-new, 2025-06-30:
Does anyone know how the solid smart plug can be turned on using the integration?

thomluther, 2025-06-30:
> Does anyone know how the solid smart plug can be turned on using the integration?

See #150

Erdkopp, 2025-07-05:
Just installed my Solarbank 3 with Anker Smart Meter. Currently in First Time Learning Phase of the Smart Mode.

Can I help you with any data?

Erdkopp, 2025-07-05:
> Potentially it could be the learning phase after Smart Mode activation, while smart mode is active mode but self consumption is executed accordingto user guide. So an admin export after first time smart mode activation will be helpful as well.
> 

[Philipp_2025-07-05_2033.zip](https://github.com/user-attachments/files/21079088/Philipp_2025-07-05_2033.zip)

As I am in this phase with my setup (first activation of smart mode and currently in learning phase), there is my export. Unfortunately I am in the "user_scene_mode 7"

So I think your thought is wrong. But maybe you'll find something else useful in my export?

thomluther, 2025-07-10:
I released 3.0.0 and hope I got the various AI status codes correctly.
They have to be distinguished between monitoring (runtime) status and training status and are displayed in different sensors, since runtime status seems to remain 0 (inactive) until the training is completed (5)
Beside that I'm closing this issue now.
Any new or missing findings with the latest release need to be tracked with separate issues or discussions.
Thanks for all your support so far to get more insights on SB3 Api structures and capabilities.

The only topic on my roadmap is to add a sensor for the 24 PV forecast data. However, that seems to be available in the Cloud only if the Smart mode is active. So I have to see how to implement that.
It will NOT be a sensor which can be used in the energy dashboard, since data availability depends on your usage mode...