thomluther/ha-anker-solix #249: BLE communication

sd8rdb, 2025-04-12:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

was there already some try to sniff the Bluetooth communication? My idea is, that I put a raspberry which emulates a phone so that there is no need to use the cloud apis. 

Just asking if there was already a initiative 

BR

### Describe the solution you'd like

Bluetooth Communication with a raspberry or other device as a proxy / phone emulator 

### Describe alternatives you've considered

Cloud connection ;)

### Additional context

No

thomluther, 2025-04-12:
Hallo, to my knowledge nobody looked into BT communication yet.

thomluther, 2025-07-20:
Thanks for the update on this. Sounds interesting if you have a BT adapter in range of the device.

sd8rdb, 2025-07-23:
Have you tried the fork or the original one? 

https://github.com/flip-dots/SolixBLE

closed 2025-09-21 by thomluther