thomluther/ha-anker-solix #250: Feature Request: Customizable min/max limits for 'System Feed-in' slider

operatorian, 2025-04-14:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Yes. When using the Anker Solix E1600 (A17C0) with a small inverter, it's possible that myself, family members, or external automations (e.g., Node-RED) unintentionally set the "System Feed-in" value too high. This can potentially overload the inverter or cause instability in my setup.

### Describe the solution you'd like

I'd like an option to define custom minimum and maximum limits directly within the integration for the "System Feed-in" slider. For instance, setting a minimum value of 150 W and a maximum value of 400 W. This ensures users or external automations cannot accidentally exceed the defined operational range.

### Describe alternatives you've considered

I've tried using YAML customization (customize.yaml) to define limits, but this doesn't work since the integration does not respect those manually defined values.

Creating a secondary slider (a slider that controls another slider) is unnecessarily complicated and not user-friendly.

### Additional context

This feature would enhance the safety and ease-of-use, especially for less technical family members or automated systems, preventing unwanted or unsafe feed-in values from being set.

Thanks for considering this improvement!

thomluther, 2025-04-14:
Hi, thanks for this request.
The limits in the integration are to match the min/max limits supported by the devices.
Going beyond those limits does not make sense.
Restricting those limits further is something you could built by yourself with HA.
You can simply create an input number helper that will be used to set the output. This input helper can be limited as you like, and you can present that in a format as you like for your dashboard.
Then create an automation that is triggered on changes of the input helper and applies them to the output preset.
Likewise, the automation must be triggered also when the output preset changes (e.g. due to schedule changes or App changes), and pick up the value from the output preset entity, but of course limit those values to the limits of your input helper.

The output preset entity then must not be exposed to the dashboard directly, but just the input helper.
There is also a sensor entity that can be used to monitor the real setting of the device.

thomluther, 2025-04-14:
The problem is that it cannot be easily put into the integration.
If the integration restricts the input, it is only restricted for the integration. The schedule or the App can still set it beyond that restriction and that may cause value errors since the output preset number entitiy must reflect what is being active in the device.

operatorian, 2025-04-14:
Ok, that makes sin. :-)

closed 2025-04-14 by operatorian