thomluther/ha-anker-solix #256: Solix2 pro new config german - english

ChrSch84, 2025-04-23:
### System Health details

.

### Integration version

2.6.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I had to reset my solix pw so I had to reconfig the adding. Now all my values / sensors show up in English. 
Problem is that mz dashboard looks for the old values and my history is gone :(

Any help thank you.

### Reproduction steps

1.change pw
2. Reconfig 
3. Wait and see english sensors
...

### Debug logs

```text
.
```

thomluther, 2025-04-23:
Hi

Actually this is no defect.
The entity IDs are generated automatically based on your User and HA Language settings and the chosen translation files for DE, FR or EN as default.
If you want back the German entity IDs, make sure to switch your HA instance to German, and do the above Reconfiguration again.
This will re-register all entities with their German Translation.
You may need to restart HA after switching the Language to make it active

ChrSch84, 2025-04-29:
Switched the Language and tried again. -> That worked. Thank you

closed 2025-04-29 by thomluther