thomluther/ha-anker-solix #258: Anker Solix 2 AC - Missing Switch for AC-Emergancy Plug?

tron10000, 2025-04-26:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I wanted to Test / Use the Emergancy Plug of the Anker Solix 2 AC System (1200W) - but it seams, that it has to be activated seperatly ..
In the anker app it's called "Netzunabhängig" (Net Indipendent)
Currently i do not find this switch in your API ? 

 
   

### Describe the solution you'd like

I would be helpful to have this switch implemented in your Integration .. 

### Describe alternatives you've considered

Alternativ it's possible to activate via the anker app 

### Additional context

-

thomluther, 2025-04-27:
Thank for the Feature Request. If someone in the community finds the endpoints and parameters to query and set this datapoint I will be happy to intergrate it. But for the time beeing this is not possible since unknown...

thomluther, 2025-05-28:
I think those features just indicate how the system should behave and what options should be provided by the App.
They are not directly settings, but flags which features are supported by the actual system and which ones are not.
Not sure what the charging and discharging flags actually stand for, but they refer to the battery.
Grid feed enabled is probably reflecting the new Switch setting that was introduced recently.

Definitely, this is just showing the active features, but not a way how you can change any device setting.

thomluther, 2026-01-29:
This was implemented and delivered with version 3.5.0

mfkfx83, 2026-02-01:
Solarbank 3: Unfortunately the switch for the "Notstromsteckdose" as well as for the LED illumination do not show up. I enabled MQTT in the integrations settings and even enabled some additional categories. Am I missing something? How do I get these switches to show up? 

[image]

Thank you very much!

thomluther, 2026-02-01:
> Solarbank 3: Unfortunately the switch for the "Notstromsteckdose" as well as for the LED illumination do not show up. I enabled MQTT in the integrations settings and even enabled some additional categories. Am I missing something? How do I get these switches to show up?

Are you sure you enabled the categories? Most users think they do, but the shown categories are EXCLUDED!

Then you also have to trigger once for realtime messages or send a status request (although a status request will not cause all message types to be send by SB3. Then after the next refresh cycle (1 min) the additional entities will be registered if values are provided in the MQTT data.

thomluther, 2026-02-01:
As mentioned in the MQTT device description, it depends on the device if they send regular messages (once every 5 minutes) and which data is contained there. If you want to see all data that is described, you may have to automate Status Requests or Realtime triggers from time to time.

mfkfx83, 2026-02-01:
Thank you so much for your support. I tried now for hours and I got it that far, that the script is asking for my credentials, but then aborting with: 

```zsh
Api  Error Cannot connect to host ankerpower-api-eu.anker.com:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1081)')] for request: POST https://ankerpower-api-eu.anker.com/passport/login
Response Text: {'ab': 'DE', 'client_secret_info': {'public_key': '.............
```

I tried this on a MacOS machine. Is it better to run it on Linux?

Thank you!

thomluther, 2026-02-01:
You can run the Export action from HA.
Or the export_system.py once you installed the library. Not idea which certificate issues you get, never saw it.

closed 2026-02-05 by thomluther