thomluther/ha-anker-solix #264: Add support for anker prime 250w

Alexandre006, 2025-04-28:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No, this is not a problem. I only want to integrate the anker prime 250w charger. Within the home assistant application.

### Describe the solution you'd like

[image]

At the moment, here's what they're getting back from the anker integration for this charger. Unfortunately, the 6 usb ports are not included in the integration.

### Describe alternatives you've considered

The anker application provides the following functions for load ports 

[image]

### Additional context

No

thomluther, 2025-04-28:
Hi @Alexandre006 thanks for your request, but I doubt this is possible.
Such stand alone devices barely report anything to the power cloud, only via the MQTT cloud where not interface exists.
You may try to explore following endpoints with your HA integration when using the main account https://github.com/thomluther/anker-solix-api/blob/2d2c2808323174cab7a42cde05f0c68e7e2319ed/api/apitypes.py#L291:
```
related to Prime charger models: 11 + 0 used => 11 total
    'mini_power/v1/app/charging/get_charging_mode_list',
    'mini_power/v1/app/charging/update_charging_mode',
    'mini_power/v1/app/charging/add_charging_mode',
    'mini_power/v1/app/charging/delete_charging_mode',
    'mini_power/v1/app/setting/set_charging_mode_status',
    'mini_power/v1/app/egg/get_easter_egg_trigger_list',
    'mini_power/v1/app/egg/add_easter_egg_trigger_record',
    'mini_power/v1/app/egg/report_easter_egg_trigger_status',
    'mini_power/v1/app/setting/get_device_setting',
    'mini_power/v1/app/power/get_day_power_data',
    'mini_power/v1/app/style/get_clock_screensavers',  # works for {'product_code': 'A2345'} => Prime charger
```

But I don't think they will report the actual power values...If you are lucky, you can find some specific settings and eventually also how those settings can be modified. You will need to 'interpret' any values when comparing them with the App.

thomluther, 2025-05-24:
@Alexandre006 , did you find any interesting data with those endpoints?  Can you provide example?

ShonP40, 2025-08-08:
@thomluther I tried some of those endpoints on my charger and here's what I got:

`mini_power/v1/app/charging/get_charging_mode_list`:
```
switch.shonp40_api_usage:
  request:
    server: https://ankerpower-api-eu.anker.com
    method: post
    endpoint: mini_power/v1/app/charging/get_charging_mode_list
    payload:
      device_sn: <redacted>
      device_model: A2345
      product_code: A2345
  response:
    code: 0
    msg: success!
    data:
      charging_mode_list: []
    trace_id: <redacted>
```

`mini_power/v1/app/egg/get_easter_egg_trigger_list`:
```
switch.shonp40_api_usage:
  request:
    server: https://ankerpower-api-eu.anker.com
    method: post
    endpoint: mini_power/v1/app/egg/get_easter_egg_trigger_list
    payload:
      device_sn: <redacted>
      device_model: A2345
      product_code: A2345
  response:
    code: 0
    msg: success!
    data:
      egg_trigger_list:
        - egg_type: 1
          trigger_time: 1734969388
          status: 1
        - egg_type: 2
          trigger_time: 1734969738
          status: 1
        - egg_type: 3
          trigger_time: 1734969851
          status: 1
        - egg_type: 4
          trigger_time: 1734969461
          status: 1
        - egg_type: 5
          trigger_time: 1734969037
          status: 1
    trace_id: <redacted>
```

`mini_power/v1/app/setting/get_device_setting`:
```
switch.shonp40_api_usage:
  request:
    server: https://ankerpower-api-eu.anker.com
    method: post
    endpoint: mini_power/v1/app/setting/get_device_setting
    payload:
      device_sn: <redacted>
      device_model: A2345
      product_code: A2345
  response:
    code: 0
    msg: success!
    data:
      device_setting:
        charging_mode_status: 1
        compatibility_status: 0
    trace_id: <redacted>
```

`mini_power/v1/app/style/get_clock_screensavers`:
```
switch.shonp40_api_usage:
  request:
    server: https://ankerpower-api-eu.anker.com
    method: post
    endpoint: mini_power/v1/app/style/get_clock_screensavers
    payload:
      device_sn: <redacted>
      device_model: A2345
      product_code: A2345
  response:
    code: 0
    msg: success!
    data:
      category:
        - id: Futuristic
          category_name: Futuristic
          list:
            - id: "948897111"
              title: Celestial
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/T5IyObNNgxhUVPH8/%E6%B0%94%E6%B3%A1.jpg
              file_crc32: "0x40914327"
              text_color: ""
              bin_url: ""
            - id: "236632206"
              title: Robotic
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/ZOXFASuOK3154M0O/%E6%9C%BA%E7%94%B2.jpg
              file_crc32: "0x756eab66"
              text_color: ""
              bin_url: ""
        - id: Cosmic
          category_name: Cosmic
          list:
            - id: "938963984"
              title: Lunar
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/dTHYjLQQJXwkoXJC/%E6%9C%88%E7%90%83.jpg
              file_crc32: "0xbd2ddca7"
              text_color: ""
              bin_url: ""
            - id: "819158638"
              title: Galactic
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/z0pv3dszhjkAYSfc/%E6%98%9F%E4%BA%91.jpg
              file_crc32: "0xdab95e74"
              text_color: ""
              bin_url: ""
            - id: "95294588"
              title: Meteorite
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/xNe6kVzxj3C1aRKh/%E6%B5%81%E6%98%9F%E9%9B%A8.jpg
              file_crc32: "0x2c1f806"
              text_color: ""
              bin_url: ""
        - id: Artistic
          category_name: Artistic
          list:
            - id: "130254227"
              title: Floral Waltz
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/dC2FoNMaEcpGz2kh/%E8%8A%B1%E4%B9%8B%E8%88%9E.jpg
              file_crc32: "0xde0cfff4"
              text_color: ""
              bin_url: ""
            - id: "347235639"
              title: Impressionist Bloom
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/P51puX6BWUwIbi7m/%E8%8E%AB%E5%A5%88%E8%8A%B1%E5%9B%AD.jpg
              file_crc32: "0x92800b00"
              text_color: ""
              bin_url: ""
        - id: Lively
          category_name: Lively
          list:
            - id: "172033515"
              title: Sweet Treats
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/dRHwN3JaCBmkEQyw/%E7%BC%A4%E7%BA%B7%E7%B3%96%E6%9E%9C.jpg
              file_crc32: "0x509eeca5"
              text_color: ""
              bin_url: ""
            - id: "496957652"
              title: Rhythmic Hue
              image_url: >-
                https://public-aiot-ore-qa.s3.dualstack.us-west-2.amazonaws.com/anker-power/public/banner/2025/03/26/iot-admin/6CF299B2kJQMth5E/%E8%B7%83%E5%8A%A8%E6%97%8B%E5%BE%8B.jpg
              file_crc32: "0xb5d0abf9"
              text_color: ""
              bin_url: ""
    trace_id: <redacted>
```

thomluther, 2025-08-08:
More questions:
The endpoints indicate that you can create charging modes and toggle them. If you don't create one, the list will be empty of course. That might be something to intergrate into HA, at least toggling between defined modes...
```
    'mini_power/v1/app/charging/get_charging_mode_list',
    'mini_power/v1/app/charging/update_charging_mode',
    'mini_power/v1/app/charging/add_charging_mode',
    'mini_power/v1/app/charging/delete_charging_mode',
    'mini_power/v1/app/setting/set_charging_mode_status',
``` 

No idea what those 'easter eggs' are!?!

For energy stats you may need deeper testing of this endpoint:
` 'mini_power/v1/app/power/get_day_power_data',`

thomluther, 2025-08-12:
There is also a query for easter egg trigger status, which should look like this, maybe based on output of trigger list?
```
endpoint: "mini_power/v1/app/egg/report_easter_egg_trigger_status"
payload: 
  device_sn: <deviceSn>
  report_time: 1734969388
  egg_type: 1
```
where egg_type is given in the list and report_time is Unix Epoche timestamp, not sure if related to the trigger time in the list...

spitfire, 2025-09-02:
> - id: Futuristic
>           category_name: Futuristic
>           list:
>             - id: "948897111"
>               title: Celestial
>               image_url: >- [...]

these sound like themes you can set on device's screen (from Anker's app):

[image]

> switch.shonp40_api_usage:
>   request:
>     server: https://ankerpower-api-eu.anker.com
>     method: post
>     endpoint: mini_power/v1/app/egg/report_easter_egg_trigger_status
>     payload:
>       device_sn: <redacted>
>       egg_type: 1
>       report_time: 1734969388
>   response:
>     code: 0
>     msg: success!
>     data: {}
>     trace_id: <redacted>

and these are easter eggs/achievements you can unlock by using the device.

[image]

I wonder if there's useful data controls somewhere?

thomluther, 2025-09-03:
There are a couple of other endpoints you have to try:
```python
    'mini_power/v1/app/charging/update_charging_mode',
    'mini_power/v1/app/charging/add_charging_mode',
    'mini_power/v1/app/charging/delete_charging_mode',
    'mini_power/v1/app/setting/set_charging_mode_status',
    'mini_power/v1/app/egg/add_easter_egg_trigger_record',
    'mini_power/v1/app/egg/report_easter_egg_trigger_status', # {"device_sn": deviceSn, "report_time": 1734969388, "egg_type": 1}
    'mini_power/v1/app/setting/set_compatibility_status',
```

thomluther, 2025-09-10:
I don't know what this charging mode is all about, but the Api calls seems to require that you must create them first, before you can update them.
The list call should show the defined ones, and as long there is no charging mode defined, you will get failures to update them.
Once you have anything in the list output, you probably also get the fields that you have to modify with the update calls.

thomluther, 2025-11-21:
Has anyone of you looked in the same issue in the Api library?
https://github.com/thomluther/anker-solix-api/issues/188

I don't think you can find much in the Api, but with MQTT messages you probably get the same values and control capability as with the App.
You just need to decode them with the MQTT monitor and describe all messages and commands for the setting changes (along with supported control options and ranges).

thomluther, 2025-11-30:
> Thanks a lot for your work, and I’m really looking forward to any progress on this!

Hi @Cyberdr3am 
Progress can only happen if someone will decode and describe your device type.
You can contribute by starting with the [mqtt_monitor tool](https://github.com/thomluther/anker-solix-api#mqtt_monitorpy) from the [Api library](https://github.com/thomluther/anker-solix-api) and follow the [decoding guidelines](https://github.com/thomluther/anker-solix-api/discussions/222)

thomluther, 2026-01-29:
Implemented some base data with version 3.5.0.
Will keep this opened if anyone describes more MQTT data or MQTT commands. I think the Api queries are not really useful, unless someone finds the queries to toggle the usage modes. But that might be possible via MQTT as well, which is easier to find with the mqtt_monitor tool.

fmranger, 2026-05-01:
This is the result from HA:

Is this helpful? 

`button.schreibtisch_250w_prime_charger_details_aktualisieren:
  device_info:
    device_sn: XXX
    is_admin: true
    owner_user_id: XXX
    device_pn: A2345
    mqtt_supported: true
    mqtt_overlay: false
    mqtt_status_request: true
    type: charger
    bt_ble_mac: XXX
    name: 250W Prime Charger
    alias: Schreibtisch 250W Prime Charger
    wifi_online: true
    wifi_name: XXX
    relate_type:
      - ble
      - wifi
    charge: false
    bws_surplus: "0"
    sw_version: v2.1.1.6
    tag: ""
    platform_tag: ""
    rssi: "-48"
    wifi_signal: "100"
    intgr_device: null
    wifi_mac: XXX
    time_zone: CET-1CEST,M3.5.0,M10.5.0/3
    group_info: null
    auto_upgrade: false
    is_ota_update: false
    ota_version: v2.1.1.6
    mqtt_data:
      sw_version: 2.1.1.6
      usbc_1_status: 1
      usbc_1_voltage: "5.088"
      usbc_1_current: "0.810"
      usbc_1_power: "4.000"
      usbc_2_status: 0
      usbc_2_voltage: "0.000"
      usbc_2_current: "0.020"
      usbc_2_power: "0.000"
      usbc_3_status: 1
      usbc_3_voltage: "9.008"
      usbc_3_current: "0.603"
      usbc_3_power: "5.400"
      usbc_4_status: 0
      usbc_4_voltage: "0.000"
      usbc_4_current: "0.020"
      usbc_4_power: "0.000"
      usba_1_status: 0
      usba_1_voltage: "0.000"
      usba_1_current: "0.000"
      usba_1_power: "0.000"
      usba_2_status: 1
      usba_2_voltage: "4.992"
      usba_2_current: "0.175"
      usba_2_power: "0.490"
      usbc_1_switch: 1
      usbc_2_switch: 1
      usbc_3_switch: 1
      usbc_4_switch: 1
      usba_switch: 1
      msg_timestamp: 1777623925
      last_update: "20`