thomluther/ha-anker-solix #273: Error rendering state template for sensor

DenisMtfl, 2025-05-23:
### System Health details

Anker Solix API not working after a time.

### Integration version

2.7.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '0f###masked###2802###masked###0695###masked###ed', 'gtoken': '0c###masked###5ef3###masked###f5'}
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2025-05-23 12:51:24.599 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_den*** data in 0.006 seconds (success: False)
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_tag: TypeError: '>' not supported between instances of 'float' and 'str'
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_stunde: TypeError: '>' not supported between instances of 'float' and 'str'
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_minute: TypeError: '>' not supported between instances of 'float' and 'str'

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
2025-05-23 12:51:24.594 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator den*** is updating sites 
2025-05-23 12:51:24.594 DEBUG (MainThread) [custom_components.anker_solix] Updating api den*** sites data
2025-05-23 12:51:24.594 DEBUG (MainThread) [custom_components.anker_solix] Getting api den*** site list
2025-05-23 12:51:24.596 DEBUG (MainThread) [custom_components.anker_solix] Loaded JSON from file /config/custom_components/anker_solix/solixapi/authcache/de###masked###l@gm###mask.json:
2025-05-23 12:51:24.596 DEBUG (MainThread) [custom_components.anker_solix] Data: {'user_id': '79###masked###960c###masked###0cdb###mas', 'email': 'de###masked###l@gm###mask', 'nick_name': 'den***', 'auth_token': '0f###masked###2802###masked###0695###masked###ed', 'token_expires_at': 1750495755, 'avatar': '', 'mac_addr': '', 'domain': '', 'ab_code': 'DE', 'token_id': 68813639, 'geo_key': 'AI###masked###j_R4###masked###Tg2m###ma', 'privilege': 0, 'phone_code': '', 'phone': '', 'phone_number': '', 'phone_code_2fa': '', 'phone_2fa': '', 'server_secret_info': {'public_key': '04c809d2534c48cfedc01dbcc2aadeb302e81f282a1584ec244afed13332ec366306cdcd5eb617c1f484ce1c88ea8455a7880d1ddd37dcd60179eccc0f689283a4'}, 'params': [], 'trust_list': None, 'fa_info': {'step': 0, 'info': ''}, 'country_code': 'DE', 'ap_cloud_user_id': '79c857209dd1f0960c8eeaef3afb550cdbc12217'}
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Cached Login for de###masked###l@gm###mask from 2025-05-22T10:49:15.811791:
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] {'user_id': '79###masked###960c###masked###0cdb###mas', 'email': 'de###masked###l@gm###mask', 'nick_name': 'den***', 'auth_token': '0f###masked###2802###masked###0695###masked###ed', 'token_expires_at': 1750495755, 'avatar': '', 'mac_addr': '', 'domain': '', 'ab_code': 'DE', 'token_id': 68813639, 'geo_key': 'AI###masked###j_R4###masked###Tg2m###ma', 'privilege': 0, 'phone_code': '', 'phone': '', 'phone_number': '', 'phone_code_2fa': '', 'phone_2fa': '', 'server_secret_info': {'public_key': '04c809d2534c48cfedc01dbcc2aadeb302e81f282a1584ec244afed13332ec366306cdcd5eb617c1f484ce1c88ea8455a7880d1ddd37dcd60179eccc0f689283a4'}, 'params': [], 'trust_list': None, 'fa_info': {'step': 0, 'info': ''}, 'country_code': 'DE', 'ap_cloud_user_id': '79c857209dd1f0960c8eeaef3afb550cdbc12217'}
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '0f###masked###2802###masked###0695###masked###ed', 'gtoken': '0c###masked###5ef3###masked###f5'}
2025-05-23 12:51:24.597 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2025-05-23 12:51:24.599 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_den*** data in 0.006 seconds (success: False)
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_tag: TypeError: '>' not supported between instances of 'float' and 'str'
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_stunde: TypeError: '>' not supported between instances of 'float' and 'str'
2025-05-23 12:51:24.916 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering state template for sensor.stromverbrauch_max_pro_minute: TypeError: '>' not supported between instances of 'float' and 'str'
```

thomluther, 2025-05-23:
Hi,
Is that related to the same DNS errors another user desribes in this discussion #270 ?
Not sure what is happening there, I don't see these type of DNS errors.
Maybe Anker is changing IPs of their cloud servers and DNS caches don't update accordingly?

There is nothing I can do about if the Anker cloud server host name is not resolvable properly by your HA instance...

DenisMtfl, 2025-05-23:
Hm okay. I thought it's because Anker is blocking my request and my HA is to often requesting the API?

thomluther, 2025-05-23:
> Hm okay. I thought it's because Anker is blocking my request and my HA is to often requesting the API?

That would be new to me. I'm pretty sure I do more requests and never saw this. If that would be the case, you would get proper response from the cloud server, But you get a connect error because DNS does not resolve the cloud server name...

DenisMtfl, 2025-05-23:
I have build an automation so I set the Input-Watt every minute. Can it be to often?

thomluther, 2025-05-23:
I don't know. If you have a shell Add On, you can try if you can ping the Anker Server or do an nslookup to see how long it taken. If you can use the App via Wifi, your IP is not blocken by Anker.