thomluther/ha-anker-solix #276: Add support for Solarbank 2 temperature

toedter, 2025-06-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The Solarbank 2 Pro gets hot (and loud) when a lot of solar power is consumed and converted. I have integrated an automation for a fan based on the solar power. It would be great if I could integrate a fan automation based on the temperature (the one that is displayed in the Anker app).

### Describe the solution you'd like

Could you provide a home assistant entity for the temperature?

### Describe alternatives you've considered

As alternative I use the solar power entity for the fan automation

### Additional context

no additional context

kitokirisaki, 2025-06-01:
See issue #26

toedter, 2025-06-01:
sorry, I have overlooked #26 and close this as duplicate.