thomluther/ha-anker-solix #277: Battery Capacity is reported as "3200 Wh"

samgeorge, 2025-06-01:
### System Health details

## System Information

version | core-2025.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1985
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.3
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 28.0 GB
disk_used | 16.1 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | green
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.5.1), Advanced SSH & Web Terminal (20.0.2), Get HACS (1.3.1), Studio Code Server (5.19.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 1
views | 3
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth6c16d3a (disabled), veth4174a71 (disabled), vethab90119 (disabled), vethf5c74c2 (disabled), veth22c6dfb (disabled), vethd56982a (disabled), veth1be22e0 (disabled), veth46bab9d (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.1.4/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth6c16d3a (), veth4174a71 (), vethab90119 (), vethf5c74c2 (), veth22c6dfb (), vethd56982a (), veth1be22e0 (), veth46bab9d ()
ipv6_addresses | lo (::1/128), end0 (fd83:17cd:5854:e64d:c4a9:d7db:e36a:4384/64, fe80::9372:61a2:7641:2a5e/64), hassio (fe80::bcfd:bfff:fe95:bb84/64), docker0 (fe80::18b1:a3ff:fe38:7aa/64), veth6c16d3a (fe80::248c:39ff:fe6c:c98c/64), veth4174a71 (fe80::e467:45ff:fe38:872c/64), vethab90119 (fe80::244b:eaff:fece:d6c/64), vethf5c74c2 (fe80::90a4:45ff:fecc:c90d/64), veth22c6dfb (fe80::c825:52ff:fe53:38e/64), vethd56982a (fe80::7c9e:d1ff:fef0:ed71/64), veth1be22e0 (fe80::c02d:4eff:fe60:14e0/64), veth46bab9d (fe80::ad:17ff:fe46:58c1/64)
announce_addresses | 192.168.1.4, fd83:17cd:5854:e64d:c4a9:d7db:e36a:4384, fe80::9372:61a2:7641:2a5e

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 28 March 2025 at 17:16
-- | --
current_recorder_run | 31 May 2025 at 21:45
estimated_db_size | 4005.52 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Integration version

2.8.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have a 1600 and a 2688 Battery installed now on my Solarbank 2 E1600 Pro, HA is reporting my capacity is 3200 Wh where it should be 4288 Wh

### Reproduction steps

Open the Solarbank 2 E1600 Pro Battery energy open the Attributes tab and Capacity is reported there.

### Debug logs

```text
N/A
```

thomluther, 2025-06-01:
**This is a known problem and cannot be fixed. This was and will also be mentioned in every release note since 2.7.0**

Neither capacity nor type of expansion packs are reported by the known Api responses.
The battery capacity entity is just calculated by the integration from SOC * Controller Battery Cap * Expansion units.
Once you intermix different capacity in expansion units, the calculation is wrong of course.
There are just 2 options:
- Remove the entity from the integration completely, then everybody who wants it has to build its own template sensor for it
- Leave the entity as is, usable for everyone that has no intermix. Those with intermix can deactivate the entity and build their own with SOC * installed total Capacity

It's as simple as that.

mike4001, 2025-06-05:
@thomluther Wouldn't it be possible that you just add the value to type in by the user in the integration configuration menu?

So we just have a field to put in for instance 4800 if we have 3x SB2 batteries or 5880 if you have 2x SB2 batteries + 1x SB3 battery.

So we would not have to create our own template entities.

mike4001, 2025-06-05:
OK I am of course no developer but it seems now you store the values 60, 10, 0,3 and 10 which I can change myself.

Could't you also store a fifth number "4800" which you use for calculating the battery capacity?

[image]

thomluther, 2025-06-05:
One more thought, since it really is not easy. The remaining energy calculation is done first in the cache, from the existing values. So initially the user input can never be available and the value would be wrong. 
That means not only the capacity would have to be 'overwritten', but also the calculation of the library or the was how this is represented in HA.
I really don't like this 'work around' to tweak Api values for proper representation in HW. Therefore HA exists and users can easily create a template if they need it. The work around probably causes more issues and complaints than it is worth the effort...

mike4001, 2025-06-05:
If you say it is not easily possible I accept this of course :-)

A template entity is a quick task for the user, this is absolutely true.

thomluther, 2025-06-05:
> Could't you also store a fifth number "4800" which you use for calculating the battery capacity?

No, that is configuration of the overall interface. What you want is a value for a certain device....how should that be mapped to users with multiple systems, various systems types, various devices or solarbanks?
Those configuration options should 'NOT' be dependent on the hierarchy of the found devices, since this may be different on various restarts or when switching accounts

mike4001, 2025-06-05:
OK, this I understand.

"My" solution would only work for 1 system and would break everything if you have more than 1 system.

thomluther, 2025-06-05:
> OK, this I understand.
> 
> "My" solution would only work for 1 system and would break everything if you have more than 1 system.

Not even then. What if the system has multiple batteries that are tracked separately like the X1?

thomluther, 2025-06-10:
I need to evaluate if restore capability of entity can be used to store custom settings in HA. After restore of entity states, they could be written as customized values to any recreated Api Cache and used in preference for calculations done in Api library. 
Not sure however if the restore entity will survive entity re-registration, which is required upon hub reconfiguration.

MarcusK66, 2025-06-30:
Anker's app displays the capacity of each battery as a percentage. If the integration could also extract these values, anyone could calculate their own remaining capacity (for different battery sizes). Everyone has the information from their system about what capacity each battery pack has.

MarcusK66, 2025-06-30:
[image]
[image]

[image]

You'll never measure the exact battery status of a system that stores data in the cloud. The measurement methods will be too inaccurate. High sample rates and cloud storage at no additional cost are contradictory. For now, the percentages for individual batteries would be sufficient for me.
Regardless, thanks for your integration (as it is)!

thomluther, 2025-07-07:
Note:
The capacity will become a new adjustable number entity with R 3.0.0.
This will also be used to correct the calculated energy, based on customized capacity and SOC.

thomluther, 2025-07-10:
Implemented with 3.0.0