thomluther/ha-anker-solix #279: no connection(log in)

TKranz75, 2025-06-02:
### System Health details

## System Information

version | core-2025.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4993
Installed Version | 2.0.5
Stage | running
Available Repositories | 1996
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 31. Dezember 2025 um 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-24.ui.nabu.casa
certificate_status | ready
instance_id | 1c08dfc47a57415cb903a291a7b89f31
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.3
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 468.7 GB
disk_used | 12.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2), Studio Code Server (5.19.3), Mosquitto broker (6.5.1), Zigbee2MQTT (2.4.0-1), eufy-security-ws (1.9.2), go2rtc (1.9.9)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 9
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), eno1 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth973a74e (disabled), veth6efabf6 (disabled), veth3a5098e (disabled), vethec73ba7 (disabled), veth1070895 (disabled), veth84c07c0 (disabled), vethe8a3bdc (disabled), veth8e52278 (disabled), vetheef7698 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), eno1 (172.16.5.20/16), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth973a74e (), veth6efabf6 (), veth3a5098e (), vethec73ba7 (), veth1070895 (), veth84c07c0 (), vethe8a3bdc (), veth8e52278 (), vetheef7698 ()
ipv6_addresses | lo (::1/128), eno1 (fd7a:5f9c:f729:0:4924:50b2:859d:13b4/64, 2001:16b8:83c8:f300:7419:43eb:8743:5f01/64, fe80::cfde:546f:6d40:428f/64), hassio (fe80::2c07:1ff:fecf:6576/64), docker0 (fe80::1087:85ff:feb4:9502/64), veth973a74e (fe80::5f:b0ff:fe72:9c19/64), veth6efabf6 (fe80::c4dd:e8ff:fea2:e631/64), veth3a5098e (fe80::70fa:4fff:fe81:4a47/64), vethec73ba7 (fe80::a854:82ff:fe2b:7e85/64), veth1070895 (fe80::b8c5:3eff:fe42:d9d5/64), veth84c07c0 (fe80::5449:ecff:fe53:acfb/64), vethe8a3bdc (fe80::a4fe:83ff:fe4b:e3b8/64), veth8e52278 (fe80::48cc:c4ff:fe98:7a54/64), vetheef7698 (fe80::3c26:d2ff:fe9a:9b94/64)
announce_addresses | 172.16.5.20, fd7a:5f9c:f729:0:4924:50b2:859d:13b4, 2001:16b8:83c8:f300:7419:43eb:8743:5f01, fe80::cfde:546f:6d40:428f

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17. Mai 2025 um 16:06
-- | --
current_recorder_run | 1. Juni 2025 um 00:15
estimated_db_size | 415.69 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Integration version

tested with 2.8.0 and 2.5.5

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I can no longer establish a connection to my Anker Solix system. I always get the error “Setup error: Found shared devices with Tino”. The problem has been going on for a week, before that it worked. I'm not sure if it has something to do with an Eufy camera, I made some changes last week.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 13:35:54 (1 Vorkommnis)
Zuletzt protokolliert: 13:35:54

Error setting up entry Tino for anker_solix: Found shared Devices with Tino
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/anker_solix/__init__.py", line 119, in async_setup_entry
    raise ConfigEntryError(
    ...<9 lines>...
    )
homeassistant.exceptions.ConfigEntryError: Found shared Devices with Tino
```

thomluther, 2025-06-02:
Hi
That means that you have at least one integration hub entry, that shares the same devices with the one you want to configure.
Try to delete the config entries and configure them new.
This will re-use old entities if they are found in your configured hub.

You cannot access the same systems/devices with an owner account and a member account in HA, this would cause duplicate unique IDs in HA since they use the SNs in the UID.
Your Eufy account has nothing to do with the Anker Power server to my knowledge

TKranz75, 2025-06-02:
Hello,
Thank you for your answer.
The main account is set up for HA and a guest account for my cell phone.
Where exactly can I delete old entries? I have already searched for this, but have not found anything in HA. I deleted all guest accounts in the Anker Solix app, but without success in HA.

thomluther, 2025-06-02:
Then you have old devices in your HA registry that have not been removed properly and now clash with the new integration entries.
You need to remove all integration entries and see under Devices and Entities if you still have something related to Anker. Then you maybe can delete those manually. Once you got rid of them, the new configuration entry should hopefully work and no longer find same devices SN existing in your HA registry.

TKranz75, 2025-06-02:
i have only one integration:

[image]

In this i have 4 "devices":

[image]

the last "Tino DE" i can't edit.

TKranz75, 2025-06-03:
Today i remove the anker solix Integration and install New. Now working fine.
Thanks for Support thomluther.