thomluther/ha-anker-solix #283: Found shared Devices with ....

imLeeat, 2025-06-05:
### System Health details

From time to time the integration fails with the error message: Einrichtungsfehler: Found shared Devices with
After restart everything is fine until this error apears again a few days/weeks later.

I am not aware of any shared devices, so how can i solve that issue?

### Integration version

2.8.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

From time to time the integration fails with the error message: Einrichtungsfehler: Found shared Devices with
After restart everything is fine until this error apears again a few days/weeks later.

I am not aware of any shared devices, so how can i solve that issue?

### Reproduction steps

1.wait until this fault happens
2. restart integration
3. wait until this fault happens
...

### Debug logs

```text
-
```

thomluther, 2025-06-05:
Hi, actually this means there is something left in your HA device registry that no longer is bound to the Hub entry.
The error tells you to which Account alias the shared devices are bound to.
Maybe you changed the name of your Account alias? This may require to rename devices and entities upon restart and eventually cause issues.
You may try to uninstall the integration and re-install it to see if that fixes the issue.

thomluther, 2025-06-05:
Pls see issue  #279 
That user had similar problems and fixed it with re-installation

closed 2025-07-13 by thomluther