thomluther/ha-anker-solix #303: Actions for Solarbank schedule not available  for Solarbank 3 Pro

tsoybe, 2025-07-10:
### System Health details

## System Information

version | core-2025.7.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.35-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4993
Installed Version | 2.0.5
Stage | startup
Available Repositories | 2100
Downloaded Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.1
agent_version | 1.7.2
docker_version | 28.3.0
disk_total | 30.8 GB
disk_used | 13.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | false
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Terminal & SSH (9.18.0), Piper (1.6.2), Whisper (2.6.0), Get HACS (1.3.1), Z-Wave JS (0.17.0), NGINX Home Assistant SSL proxy (3.13.0), Let's Encrypt (5.4.9), Speech-to-Phrase (1.4.1), Log Viewer (0.17.1), FHEM (0.1.38), MQTT-anonymous (0.1.33), Advanced SSH & Web Terminal (21.0.2), Miniflux (0.2.26), Music Assistant Server (2.5.5), YT Music PO Token Generator (1.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 7
views | 8
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), enp0s19 (disabled), docker0 (disabled), hassio (disabled), vethb72dfbe (disabled), veth2e527f9 (disabled), veth03cf054 (disabled), vethfb691a4 (disabled), veth728a286 (disabled), veth394719e (disabled), veth579ab72 (disabled), veth8dfc72d (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 (10.60.0.21/16), enp0s19 (10.80.10.10/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), vethb72dfbe (), veth2e527f9 (), veth03cf054 (), vethfb691a4 (), veth728a286 (), veth394719e (), veth579ab72 (), veth8dfc72d ()
ipv6_addresses | lo (::1/128), enp0s18 (fe80::a793:8c6f:da0d:75dc/64), enp0s19 (fe80::7d08:a9c3:2b30:7320/64), docker0 (fe80::68d2:7cff:feeb:490f/64), hassio (fe80::60f3:98ff:fe13:be93/64), vethb72dfbe (fe80::349b:8aff:fe30:2226/64), veth2e527f9 (fe80::30c2:43ff:fe5b:3a9b/64), veth03cf054 (fe80::783c:f3ff:fe8a:95e8/64), vethfb691a4 (fe80::b4a7:2fff:fe73:a1d/64), veth728a286 (fe80::dc05:feff:fe30:a17a/64), veth394719e (fe80::fce7:b7ff:fec6:814f/64), veth579ab72 (fe80::b4d0:21ff:fecb:383/64), veth8dfc72d (fe80::8ceb:c3ff:fea1:e5f5/64)
announce_addresses | 10.60.0.21, fe80::a793:8c6f:da0d:75dc

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 24, 2025 at 6:45 PM
-- | --
current_recorder_run | July 10, 2025 at 10:34 PM
estimated_db_size | 769.23 MiB
database_engine | mysql
database_version | 10.11.13

</details>

### Integration version

3.0.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The actions:

* Clear the Solarbank schedule
* Request actual Solarbank schedule
* Set a new Solarbank Schedule
* Update Solarbank Schedule

can't used with  Solarbank 3 E2700 Pro (A17C5) 

Some other Actions works.

### Reproduction steps

1.Developer tools --> Actions
2. Choose one of the Actions
3. Can't choose the device

If I choose e.g. the action "Api request", I can choose my device.

### Debug logs

```text
n/a
```

thomluther, 2025-07-11:
Hi, do you use the owner account?
You can use the System export action and send me the zip file to review what is going on.

tsoybe, 2025-07-12:
I changed to the owner account and now it's works.

Thanks for your work.

closed 2025-07-12 by tsoybe