thomluther/ha-anker-solix #312: Update documentation if cloud account can be used persitently in parallel on multiple devices

thomluther, 2025-07-28:
### System Health details

N/A

### Integration version

3.1.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

As of end of July 2025, it seems Anker relaxed the policies for allowed active tokens per account.
Previously only one active token was allowed, which prevented that same account could be shared in parallel on multiple devices.
This seems now possible without logout on other devices.

### Reproduction steps

1. Use the owner account in the Anker mobile app
2. Configure the owner account also in the HA integration HUB
3. They can be used in parallel without previously seen kickoffs due to deleted token
...

### Debug logs

```text
N/A
```

domhaas, 2025-07-29:
Can confirm: works perfectly.

domhaas, 2025-07-29:
> > Can confirm: works perfectly.
> 
> Indeed it does, but lets wait whether it will still work in a couple of weeks after more cloud updates.., I'm afraid this could just be a bug due to cloud changes....who knows.

Received today an iOS-Update of the Anker-App with mentioning of logging in with multiple devices with the same account. So I don't think this is a bug. But let's see.

thomluther, 2025-07-29:
> Received today an iOS-Update of the Anker-App with mentioning of logging in with multiple devices with the same account. So I don't think this is a bug. But let's see.

Well then it is an official change and will probably be announced with the Android version as well.
BTW. This is independent of the client or app, it was the cloud server that kicked of previous tokens once the account logged in from any other client. Only one active token was possible in the past, now they seem to stop deletion of token and let them just expire.

thomluther, 2025-08-03:
Question is why you got login errors, so that they could exceed the limit, whatever the limit is...