thomluther/ha-anker-solix #318: After Update to 3.1.1 and changing account to the main Anker Account - wrong Values and new HA-IDs

RamsukKockoi, 2025-08-05:
### System Health details

The system is full OK, I reverted to an Backup.

### Integration version

actual 3.1.0
Issue Version 3.1.1

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I Updated to Version 3.1.1, yesterday. 
I changed from the "read Account" to the Main Anker account, to connect to the api. After a while I noticed a risen SOC on my SB2, about 20% more, than before an than shown in the app.

And the SB2 DC Output was about half of the rate, the SB2 was putting out (could see it, because a Shelly Plug measured another Value and the app showed something similar).

So I went back to read Only accout, just to see, now I had a "new" SB2 in HAS. That truely scared me out of my pants. Luckily "Autobackup" Integration saved my live, I went back an had just lost about 2 Hours.

If Neeeded, I am willing to repeat all the steps I did, to produce the Issue. Let me know, what you need to analyze.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Not yet needed, I went back to an older Backup
```

### Diagnostics dump

Not yet needed, I went back to an older Backup

thomluther, 2025-08-05:
Hi, I'm afraid there is nothing I can do about that. 
Typically if you reconfigure your account login, or you delete and Hub and reconfigure a new one, the old entities will be re-used.
Exception here is that you either changed your device names or the language setting in Home Assistant between those reconfigurations, because the device names will automatically be used + the entity translations to compose the entity names.
However, in backend the unique_id is the same, which is using the serial number. So typically the old entities are restored if the same unique ID is composed.

> After a while I noticed a risen SOC on my SB2, about 20% more, than before an than shown in the app.

That is nothing the integration would do, it does not change things automatically. If you saw it in the app, that it occurred by the device or your system. The integration data is received only in 5 minute intervals from the cloud, the SB2 does not update the cloud more often. There is real time data in the cloud and the integration only reports what the cloud reports.
There was absolutely no change in that regard since the previous version.

So to sum up, I don't know what issue you had, what went wrong and how I could help.

RamsukKockoi, 2025-08-05:
Hey,

I know, that it is only actualised about every 5 minutes. The Problem is, that I saw those values for a long time. At least for about 30-45 Minutes, for first I did not realize, they were Wrong. We had a TV running, a Gaming PC and a Gaming Notebook, together more than 800 Watts. So the SB2 was on Powerlimit, while the data in HomeAssistant was about 350 to 400 Watts.

When I changed forward and back, I did not delete the hub, I just changed username and password. Could that be a problem. No changes to language or device names.

I will give it another try tomorrow, as today I do not think we will have enough sun for a good test.

RamsukKockoi, 2025-08-07:
Enough Sun today. Here some Screenshots.

Pre Update
PostUpdate "RO account"
Post Update "WR account, main"
Postupdate back to RO

[image]
[image]
[image]

RamsukKockoi, 2025-08-07:
I did the same with the old Version (3.1.0) of the integration and Tadaa, the same behavior!

Weird, do you have any ideas?

thomluther, 2025-08-07:
All this entity_id naming is being done automatically by core to avoid duplicate entity_ids. For that reason I also do not recommend to use integration entities directly in Energy dashboard, since a simple HA language setting change or intergration change can cause entity_id change and screw up you energy history. When you use a template entity in the energy dashboard (for example an integral helper that accumulates energy over a power entity of the integration), then you won't have such issues. If the power entity id changes, you just need to update the source entitiy of your integral helper.

thomluther, 2025-08-07:
If you want to compare the numbers from the cloud, you can use the action System Info for your system. You may use the merge cache option to see more background data, that is available in the Api cache for your system.
Those numbers will be extracted for the entities of the system and the devices in the system.

thomluther, 2025-08-07:
> But why does the SoC change?

Because your old dashboard entity_id is now user by another device entity?

RamsukKockoi, 2025-08-07:
Nope its the same entity. It also counts new percentages on top, that I also see in the app.
I will do some furhter investigations.

thomluther, 2025-08-13:
Hi @RamsukKockoi , if you could fix your entity naming issues, please close this issue again.
Thx