thomluther/ha-anker-solix #319: Sensor with class total_increasing is reporting negative value

ReneNulschDE, 2025-08-07:
### System Health details

## System Information

version | core-2025.9.0.dev0
-- | --
installation_type | Home Assistant Core
dev | true
hassio | false
docker | false
container_arch | null
user | rene
virtualenv | true
python_version | 3.13.2
os_name | Linux
os_version | 6.14.0-27-generic
arch | x86_64
timezone | Europe/Berlin
config_dir | /home/rene/.homeassistant

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4980
Installed Version | 2.0.5
Stage | running
Available Repositories | 2129
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 21, 2025 at 1:00 AM
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | false
remote_connected | false
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | false
remote_server | eu-central-1-1.ui.nabu.casa
certificate_status | ready
instance_id | 2f12bad18e8d4df8a167172255e302c5
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Link2Home</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 8
views | 14
mode | storage

</details>

<details><summary>MercedesME 2020</summary>

api_endpoint_reachable | ok
-- | --
websocket_connection_state | connected
cars_connected | 1
version | v0.29.1

</details>

<details><summary>MySmartBike</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Network Configuration</summary>
removed
</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 8, 2025 at 4:32 PM
-- | --
current_recorder_run | August 5, 2025 at 10:57 AM
estimated_db_size | 10779.17 MiB
database_engine | sqlite
database_version | 3.46.1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Integration version

3.1.1

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

HA is reporting multiple warnings like:

`Entity sensor.system_home_daily_solar_pv_3 from integration anker_solix has state class total_increasing, but its state is negative. Triggered by state -0.01 with last_updated set to 2025-08-06T05:32:22.479075+00:00. Please create a bug report at https://github.com/thomluther/ha-anker-solix/issues`
or
`Entity sensor.system_home_daily_solar_pv_3 from integration anker_solix has state class total_increasing, but its state is not strictly increasing. Triggered by state 0.20 (0.21) with last_updated set to 2025-08-05T10:59:14.480622+00:00. Please create a bug report at https://github.com/thomluther/ha-anker-solix/issues`

I assume that the reason is a conversation from float to int... (maybe a round up could help)

### Reproduction steps

No clear repo structure. Happens from time to time.

### Debug logs

```text
n/a
```

### Diagnostics dump

Diagnostic dump is not available

[image]

ReneNulschDE, 2025-08-07:
OK. I have enabled the debug log and will share the details as soon as I see the warning again. Thx

thomluther, 2025-08-07:
> OK. I have enabled the debug log and will share the details as soon as I see the warning again.

Be warned that this will cause massive logging since EACH Api query is logged. So the file will be huge. Debug Logging is more intended for an immediate recreate of the problem, so I would not run it for multiple hours...

ReneNulschDE, 2025-08-09:
Here is a debug log. The sensor "system_home_daily_solar_pv_1" is changing from "1.28" to "1.29" and then back to "1.28".

[anker.log](https://github.com/user-attachments/files/21698158/anker.log)

Thx

thomluther, 2025-08-10:
Here are the 2 subsequent requests showing that the cloud reported value goes backwards from 1.29 to 1.28.
```
2025-08-08 19:56:38.895 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/energy_analysis
2025-08-08 19:56:38.895 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'site_id': '<redacted>', 'device_sn': '<redacted>', 'type': 'week', 'start_time': '2025-08-08', 'device_type': 'solar_production_pv1', 'end_time': '2025-08-08'}
2025-08-08 19:56:39.215 DEBUG (MainThread) [custom_components.anker_solix] Api ank*** request post https://ankerpower-api-eu.anker.com/power_service/v1/site/energy_analysis response received
2025-08-08 19:56:39.215 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'power': [{'time': '2025-08-08', 'value': '1.29', 'rods': None}], 'charge_trend': None, 'charge_level': None, 'forecast_trend': None, 'power_unit': 'wh', 'charge_total': '0.82', 'charge_unit': 'kwh', 'discharge_total': '', 'discharge_unit': 'kwh', 'charging_pre': '', 'electricity_pre': '', 'others_pre': '', 'statistics': None, 'battery_discharging_total': '', 'solar_to_grid_total': '0.04', 'grid_to_home_total': '', 'ac_out_put_total': '', 'home_usage_total': '', 'solar_total': '4.87', 'trend_unit': 'kwh', 'battery_to_home_total': '', 'smart_plug_info': None, 'local_time': '', 'grid_to_battery_total': '', 'grid_imported_total': '', 'solar_to_battery_total': '0.82', 'solar_to_home_total': '4.01', 'forecast_total': '', 'sop_grid_to_battery_total': '', 'third_solar_to_battery_total': '', 'pps_list': None, 'total_energy_unit': ''}, 'trace_id': 'ecb2748ed6a9c68f809e05448204e4a4'}

2025-08-08 20:06:45.847 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/energy_analysis
2025-08-08 20:06:45.847 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'site_id': '<redacted>', 'device_sn': '<redacted>', 'type': 'week', 'start_time': '2025-08-08', 'device_type': 'solar_production_pv1', 'end_time': '2025-08-08'}
2025-08-08 20:06:46.164 DEBUG (MainThread) [custom_components.anker_solix] Api ank*** request post https://ankerpower-api-eu.anker.com/power_service/v1/site/energy_analysis response received
2025-08-08 20:06:46.164 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'power': [{'time': '2025-08-08', 'value': '1.28', 'rods': None}], 'charge_trend': None, 'charge_level': None, 'forecast_trend': None, 'power_unit': 'wh', 'charge_total': '0.82', 'charge_unit': 'kwh', 'discharge_total': '', 'discharge_unit': 'kwh', 'charging_pre': '', 'electricity_pre': '', 'others_pre': '', 'statistics': None, 'battery_discharging_total': '', 'solar_to_grid_total': '0.04', 'grid_to_home_total': '', 'ac_out_put_total': '', 'home_usage_total': '', 'solar_total': '4.89', 'trend_unit': 'kwh', 'battery_to_home_total': '', 'smart_plug_info': None, 'local_time': '', 'grid_to_battery_total': '', 'grid_imported_total': '', 'solar_to_battery_total': '0.82', 'solar_to_home_total': '4.03', 'forecast_total': '', 'sop_grid_to_battery_total': '', 'third_solar_to_battery_total': '', 'pps_list': None, 'total_energy_unit': ''}, 'trace_id': 'a7866841c66410e25cc447e4890e905d'}
```

Actually I don't know what I could do about that. I could ignore in general a smaller value if it is in 0.01 range and new value is not 0, but that could also mask a restart of real values next day, if for example the value was 0.02 and next day value starts with 0.01.
Honestly I don't know either if that is recognized by HA as a real reset of the total increasing entitiy, if the new value is not 0. But actually it could be the first energy value of the next day after daily reset.

thomluther, 2025-08-22:
Single decrement step now ignored for new value, if value is > 0.
That should hopefully reduce the warnings if just caused by rounding issues in the cloud (0.01 kWh decrease)
Implemented with 3.2.0