thomluther/ha-anker-solix #328: Add support changing Grid Connected Output

JurgenVM, 2025-08-17:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No, it’s not a problem, but it would be very useful to be able to set the output power of the battery from Home Assistant. When my hybrid vehicle is charging using solar panels, I don’t want the battery to be discharged for the car — only for the house. My consumption mode is set to self-consumption.

### Describe the solution you'd like

The ability to change the battery output power in Home Assistant, similar to how it can be adjusted in the mobile app.

### Describe alternatives you've considered

I’ve considered using custom mode, but that would require a lot of API requests to the cloud, which isn’t efficient.

### Additional context

[image]

thomluther, 2025-08-17:
Hi, what device are you talking about?
The screenshot does not look familiar to me, so I think it is a device with unknown control via the Api.
You will have to test various endpoints and queries to verify how you can adjust that setting, if possible at all.
Once you found a workable solution with endpoint and parameter format, I can add it to the Api library and integration.
But as long as nobody can tell me how the Api works with your system type, this will be a dead end.

thomluther, 2025-08-17:
If that is a solarbank device, then the setting you want is a device setting that cannot be changed via cloud API. This is similar to other device settings on that page, see #258