thomluther/ha-anker-solix #338: Anker Solix SB1 not longer delivering negative values since last update

RamsukKockoi, 2025-08-24:
### System Health details

Sysytem Health may not be relevant

### Integration version

3.2.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Since Update to 3.2.0 discharging my SB1 does not longer provide negative values. That leads to negative "gray consumption" and wrong batterie states.

I am taking Battery_Power and do put it into 2 Sensors, one is counting positive values one counts negative values. It seems negative values are no longer delivered.

Could you please checkt it? It my be just a little bug.

### Reproduction steps

1. charge batteries
2. dicharge batteries

### Debug logs

```text
As soon as you need them, I think actually they are not needed,
```

### Diagnostics dump

As soon as you ask for it

Perrylicious, 2025-08-24:
+1 on this, same issue

thomluther, 2025-08-24:
Fixed with release 3.2.1

closed 2025-08-24 by thomluther