thomluther/ha-anker-solix #353: Entities not available every appr. 30..40 minutes

bennipi1287, 2025-09-10:
### System Health details

## System Information

version | core-2025.9.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.12.43-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2194
Downloaded Repositories | 41

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>AccuWeather</summary>

error | failed to load: unknown
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 24. Oktober 2025 um 02:00
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-31.ui.nabu.casa
certificate_status | ready
instance_id | 3c12e0fdde554c498b3f74d518a48560
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

[... 108 lines cut ...]
2025-09-10 10:07:04.058 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'content-type': 'application/json', 'model-type': 'DESKTOP', 'app-name': 'anker_power', 'os-type': 'android', 'country': 'DE', 'timezone': 'GMT+02:00', 'gtoken': 'cb###masked###f620###masked###ff', 'x-auth-token': 'a7###masked###9fcd###masked###b643###masked###e2'}
2025-09-10 10:07:04.058 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {}
2025-09-10 10:07:05.362 ERROR (MainThread) [custom_components.anker_solix] Error fetching anker_solix_luk*** data: Api Connection Error: Cannot connect to host ankerpower-api-eu.anker.com:443 ssl:default [Timeout while contacting DNS servers]
2025-09-10 10:07:05.362 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_luk*** data in 1.306 seconds (success: False)
2025-09-10 10:08:05.056 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator luk*** is updating sites 
2025-09-10 10:08:05.057 DEBUG (MainThread) [custom_components.anker_solix] Updating api luk*** sites data
2025-09-10 10:08:05.057 DEBUG (MainThread) [custom_components.anker_solix] Getting api luk*** site list
2025-09-10 10:08:05.057 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list
2025-09-10 10:08:05.057 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'content-type': 'application/json', 'model-type': 'DESKTOP', 'app-name': 'anker_power', 'os-type': 'android', 'country': 'DE', 'timezone': 'GMT+02:00', 'gtoken': 'cb###masked###f620###masked###ff', 'x-auth-token': 'a7###masked###9fcd###masked###b643###masked###e2'}
2025-09-10 10:08:05.057 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {}
2025-09-10 10:08:05.160 DEBUG (MainThread) [custom_components.anker_solix] Api luk*** request post https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list response received
2025-09-10 10:08:05.160 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 10 Sep 2025 08:08:05 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '403', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': 'a4206b263f12d2eda06803d8364dac89', 'Traceparent': '00-a4206b263f12d2eda06803d8364dac89-e599f474e37e7988-00', 'Server': 'APISIX')>
2025-09-10 10:08:05.160 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'site_list': [{'site_id': '6f9ad912-7473-4d19-a60b-94b467c1c058', 'site_name': 'Solarbank Garage', 'site_img': '', 'device_type_list': [3, 6], 'ms_type': 2, 'power_site_type': 12, 'is_allow_delete': True, 'support_device_models': ['A17C5', 'A17X7', 'SHEM3', 'SHEMP3', 'A17X8', 'SHPPS', 'AE1R0'], 'current_site_device_models': ['SHEM3', 'A17C5']}]}, 'trace_id': 'a4206b263f12d2eda06803d8364dac89'}
2025-09-10 10:08:05.160 DEBUG (MainThread) [custom_components.anker_solix] Getting api luk*** scene info for site
2025-09-10 10:08:05.161 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_scen_info
2025-09-10 10:08:05.161 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'content-type': 'application/json', 'model-type': 'DESKTOP', 'app-name': 'anker_power', 'os-type': 'android', 'country': 'DE', 'timezone': 'GMT+02:00', 'gtoken': 'cb###masked###f620###masked###ff', 'x-auth-token': 'a7###masked###9fcd###masked###b643###masked###e2'}
2025-09-10 10:08:05.161 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'site_id': '6f9ad912-7473-4d19-a60b-94b467c1c058'}
2025-09-10 10:08:05.531 DEBUG (MainThread) [custom_components.anker_solix] Api luk*** request post https://ankerpower-api-eu.anker.com/power_service/v1/site/get_scen_info response received
2025-09-10 10:08:05.531 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 10 Sep 2025 08:08:05 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': 'a066b193a80f563d558811cfae0fc0f3', 'Traceparent': '00-a066b193a80f563d558811cfae0fc0f3-c8a4becbc9edd42f-00', 'Server': 'APISIX')>
2025-09-10 10:08:05.532 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'home_info': {'home_name': 'Home', 'home_img': '', 'charging_power': '0.00', 'power_unit': 'W'}, 'solar_list': [], 'pps_info': {'pps_list': [], 'total_charging_power': '0.00', 'power_unit': 'W', 'total_battery_power': '0.00', 'updated_time': '', 'pps_status': 0}, 'statistics': [{'type': '1', 'total': '689.04', 'unit': 'kWh'}, {'type': '2', 'total': '598.74', 'unit': 'kg'}, {'type': '3', 'total': '158.97', 'unit': '€'}], 'topology_type': '1', 'solarbank_info': {'solarbank_list': [{'device_pn': 'A17C5', 'device_sn': 'APCDJES0F15300019', 'device_name': 'Solarbank 3 E2700 Pro', 'device_img': 'https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2025/04/15/iot-admin/6SO8wjMetOwT8PaH/picl_A17C5_normal.png', 'battery_power': '10', 'bind_site_status': '', 'charging_power': '301', 'power_unit': 'W', 'charging_status': '0', 'status': '1', 'wireless_type': '1', 'main_version': '', 'photovoltaic_power': '301', 'output_power': '301', 'create_time': 1749118644, 'set_load_power': '', 'sub_package_num': 0, 'output_cutoff_data': 0, 'is_display': True, 'bat_charge_power': '0', 'pv_name': {'micro_inverter_name': '', 'pv1_name': 'PV1', 'pv2_name': 'PV2', 'pv3_name': 'PV3', 'pv4_name': 'PV4'}, 'pv_power': None, 'feature_switch': {'0w_feed_v2': True, 'enable_parallel': True, 'enable_timeslot': True, 'exceed_power': False, 'heating': False, 'plug_switch_report': True, 'show_third_party_pv_panel': False, 'support_p1_meter': True}, 'heating_power': '0', 'bat_discharge_power': '0', 'err_code': 0}], 'total_charging_power': '0', 'power_unit': 'W', 'charging_status': '0', 'total_battery_power': '0.10', 'updated_time': '2025-09-10 10:07:54', 'total_photovoltaic_power': '301', 'total_output_power': '301.00', 'display_set_power': False, 'battery_discharge_power': '0', 'ac_power': '0', 'to_home_load': '0', 'is_display_data': True, 'solar_power_1': '69', 'solar_power_2': '66', 'solar_power_3': '85', 'solar_power_4': '81', 'other_input_power': '0', 'micro_inverter_power': '166', 'micro_inverter_power_limit': '0', 'micro_inverter_low_power_limit': '0', 'grid_to_battery_power': '0', 'pei_heating_power': '0', 'backup_info': {'start_time': 0, 'end_time': 0, 'full_time': 0}, 'use_time_excute_info': {'current_band': '0'}}, 'retain_load': '200W', 'scene_mode': 1, 'home_load_power': '424', 'updated_time': '01-01-0001 00:00:00', 'power_site_type': 12, 'site_id': '6f9ad912-7473-4d19-a60b-94b467c1c058', 'powerpanel_list': [], 'grid_info': {'grid_list': [{'device_pn': 'SHEM3', 'device_sn': '79530342957156', 'device_name': 'Stromzähler', 'device_img': '', 'create_time': 1749119605, 'status': '1', 'bind_site_status': '', 'wireless_type': ''}], 'photovoltaic_to_grid_power': '0', 'grid [... cut ...] {'date': '2025-09-10', 'country_info': [{'country': 'DE', 'company_info': [{'company': 'Nordpool', 'area_info': [{'area': 'GER', 'area_name': 'GER'}]}]}]}, 'energy_poll_time': '2025-09-10 10:03:02', 'energy_poll_seconds': 2.354}}
2025-09-10 10:08:05.542 INFO (MainThread) [custom_components.anker_solix] Fetching anker_solix_luk*** data recovered
2025-09-10 10:08:05.542 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_luk*** data in 0.486 seconds (success: True)
```

bennipi1287, 2025-09-10:
Don't know if Anker Cloud is the problem. The app is working at the same time when the integration fails.

thomluther, 2025-09-10:
> Don't know if Anker Cloud is the problem. The app is working at the same time when the integration fails.

The App does also use the MQTT Cloud, which maybe has no connection issues

thomluther, 2025-09-10:
It does not help if everybody keeps posting to have the same issue, Since it will be mostlikely everybody if the cloud is not available.

[image]

I see it as well, and the integration recovers automatically once cloud is reachable again. It happened more often in the past around new product releases.
Thats the way it is...

Wolfi311, 2025-09-10:
Yes, that wouldn't be so tragic.
But the Energy Board generates 20 or more incorrect values (charge/discharge) that have to be corrected in the statistics every day.
Very tedious.

Wouldn't it be possible to intercept such incorrect values in the integration?  E.g.  Value >  3000Wh invalid or 0?

Or what other options are there  to intercept them?

Best regards,
Wolfgang

thomluther, 2025-09-14:
I think it is a sporadic cloud server issue. Here is an example during a system export.
One standard request in the middle timed out after 30 seconds with following error, the remaining queries worked again:
```shell
ERROR: Api Request Error: 502, message='Bad Gateway', url='https://ankerpower-api-eu.anker.com/power_service/v1/product_categories'
ERROR: Response Text: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
</body>
</html>

ERROR: Api: Mic***, Method: GET, Endpoint: power_service/v1/product_categories, Payload: {}
<class 'api.errors.ConnectError'>: (502) Anker Api Error: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
</body>
</html>
```

So a different integration version won't change much. You may see less of these connection errors if you reduce the overall queries, e.g. exclude the device types and categories you are not interested in. The energy stats need most of the queries.

thomluther, 2025-09-15:
This DNS server error is different, but also occured to a couple of people, this is typically longer and retry may not succeed. I can add that to the list as well, but it won't help.
By circumstance I just got 502 error and a retry was done, which again got 502. So if the web servers are busy they are busy

Ash2k80, 2025-09-15:
Okay, I understand, but strangely enough, the problem only occurs with this integration, and I haven't changed anything in my setup in the last few months. It runs in a VM under Proxmox, so maybe that's where the problem lies, but I don't know enough about it to know which direction to go in.

K-e-l-v-i-n-12, 2025-09-15:
Yes, I could definitely imagine that.

I have now used anker-solix-api for further testing. Here, I also regularly received the ‘502 Bad Gateway’ error. It usually worked after the third attempt. Sometimes, however, more attempts were necessary:

`        attempt = 1
        
        while True:
            try:
                console.info(f"get_message_unread... (attempt {attempt})")
                _out(await myapi.apisession.request("get", apitypes.API_ENDPOINTS["get_message_unread"], json={}))
                attempt = 1
                await asyncio.sleep(10)

            except Exception as e:
                console.error(f"error (attempt {attempt}): {e}")
                attempt += 1
                continue`

K-e-l-v-i-n-12, 2025-09-15:
> Unfortunately my error message is not very specific, but today it is very frequently occurring... I have the feeling that some part in the logging is missing. There should be some more details after the colon, right? I attached the debug log below. It looks as if the timeout occurs when sending an empty request body. Is this intentionally? Can you reproduce this or verify that it occurs at the same request for you?

The attached log is relatively informative. The last request is "https://ankerpower-api-eu.anker.com/power_service/v1/get_message_unread". There is no response from the server within the defined time, which aborts the request. If the timeout were higher, the 502 error I described would most likely appear.

thomluther, 2025-09-16:
This really seems to be a temporary issue with the unread messages query, at least for me.
For testing I also commented the corresponding line in the poller.py module and haven't had this issue since then. However, this will remove the entity completely.
I'm now testing another change that you can apply to your apibase.py module. Use VSC or File Editor add on to modify this in your installation. Afterwards you need to restart HA to activate changed integration code.

[image]

In line 16 add ClientError to imports
```python
from aiohttp import ClientError, ClientSession
```
In get_message_unread method catch ClientError from the request to basically ignore them for this endpoint. Search for 'unread' and you should find the place (replace line 791-793 with this block)
```python
            # Ignore timeouts or other errors from endpoint
            try:
                resp = await self.apisession.request(
                    "get", API_ENDPOINTS["get_message_unread"]
                )
            except ClientError:
                resp = {}
```

This will prevent the poller to raise Client error if only this endpoint has issues and you still get the entity if it works on initial polls. (refresh may then skip updates of entity if there are client errors during poll).

This may not catch all type of errors, but only the basic client errors raised by the aiohttp module. Not sure if that covers all type of Timeouts that may arise. It will catch 502 Gateway errors, but no DNS errors that were reported here.

RamsukKockoi, 2025-09-16:
Your workaround leads me to infinite timeouts, reverted it.
Maybe I did sometihing wrong, but i do not think so.

I am going to just live with it, I do not see any disadvantage from the errors.

Here is what I did, may you can see a mistake:

from pathlib import Path
from typing import Any

from aiohttp import ClientError, ClientSession

from .apitypes import (
    API_ENDPOINTS,
    API_FILEPREFIXES,

----

       """
        if fromFile:
            resp = await self.apisession.loadFromFile(
                Path(self.testDir()) / f"{API_FILEPREFIXES['get_message_unread']}.json"
            )
        else:
            # Ignore timeouts or other errors from endpoint
            try:
                resp = await self.apisession.request(
                    "get", API_ENDPOINTS["get_message_unread"]
                )
            except ClientError:
                resp = {}
        data = resp.get("data") or {}
        # New method: Save unread msg flag in account dictionary

Wolfi311, 2025-09-16:
I have implemented the changes at 12:00 p.m.

https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3296652517
https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3297080414

No interruptions to the cloud have been recorded since then.

[image]

Best regards,
Wolfgang

DenisMtfl, 2025-09-16:
> I have implemented the changes at 12:00 p.m.
> 
> [#353 (comment)](https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3296652517) [#353 (comment)](https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3297080414)
> 
> No interruptions to the cloud have been recorded since then.
> 
> [image]
> 
> Best regards, Wolfgang

Cool, but where is the code?

thomluther, 2025-09-16:
> Cool, but where is the code?

You have to code it in your installation. Either [this way](https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3296652517) keeping the message entity updated (if there is a response), or [the other way](https://github.com/thomluther/ha-anker-solix/issues/353#issuecomment-3297080414) (skipping the message entity updates).

K-e-l-v-i-n-12, 2025-09-17:
Thank you very much for the quick reply. Works wonderfully. 

Perhaps the ClientError and TimeoutError should be logged in future releases. Presumably a warning instead of an error would be sufficient here.

thomluther, 2025-09-17:
Interestingly something seems to be odd on the cloud server for the prefix of this endpoint.
Typically there is an error if the endpoint syntax is wrong. But on this endpoint, while the get_message* is used, it may result in an timeout before the server can reject the request due to invalid syntax (url does not exist)....
It appears to be an issue in the cloud service software how endpoint/urls are being parsed? But its weird that it is a temporary issue, however still quite frequent.

Lokheizer, 2025-09-17:
> 3.2.4 works without any problems! Perfect work! Danke auch von mir für deine tolle Arbeit!
Same here - great work! @thomluther

K-e-l-v-i-n-12, 2025-09-18:
The error message posted by @bennipi1287 (at least the 504 error) comes from the API gateway and is probably a temporary failure of the underlying services. I consider logging to be useful, as this is a real malfunction.

thomluther, 2025-09-19:
I will add 504 to the retry errors as well, so a short temporary issues may be covered.

thomluther, 2025-09-21:
Enhancements implemented with 3.3.0.
If anyone finds new http codes that result in temporary server issues, please reopen with the error code.

closed 2025-09-21 by thomluther

thomluther, 2025-09-22:
Given that there are so many cloud issues currently, I wonder whether I should reduce the severity of the log messages however...
I could change the retry message from Warning to Info only...
But failed retry should log an Error like any other failed request error type. This is part of the Api library and I don't think it makes sense to degrade a request error to a warning only.

thomluther, 2025-09-24:
I'm testing now following changes:
- Retry log entry is now Info instead of Warning. It can be seen only when looking at core live logs and if logger level is Info
- All requests now make only single error log entry if something went wrong
  - They still raise the error and methods catching the exception may additionally log an error if the exception is not ignored
- Timeout was made configurable in Hub options
- New Category that can be excluded: Account Information
  - This will prevent the get_message_unread and get_product requests completely, so expected errors can be avoided while the cloud issues exist.
  - Exclusion will remove the Unread Messages sensor. It may also cause no meaningful device names for member systems, since those report only the alias (custom name), which will also be used as device (model) name by default

Following is what you can see now in the logs for an endpoint failure during poll:
```
2025-09-24 15:43:50.467 INFO (MainThread) [custom_components.anker_solix] Http error 'Timeout Error', retrying request of api tho*** after delay of 2 seconds for endpoint: power_service/v1/get_message_unread
2025-09-24 15:44:01.469 ERROR (MainThread) [custom_components.anker_solix] Api tho*** Error  for request: GET https://ankerpower-api-eu.anker.com/power_service/v1/get_message_unread
Response Text: Timeout Error
```