thomluther/ha-anker-solix #354: AC charging power setting

Jorre05, 2025-09-10:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Since the latest Solarbank 3 firmware update (v1.0.2.6) I can configure the AC charging power. Before this was fixed at 1200W, now I can select between 0 and 1200W in 100W increments.

Since I have no solar panels connected to my Solarbank 3 I use the backup option to store excess solar energy from my SMA solar plant. During cloudy days there is not always 1200W surplus to store for later use.

Setting the AC charging power would enable me to store excess solar energy on cloudy days.

### Describe the solution you'd like

Set AC charging power to a value between 0 and 1200W

### Describe alternatives you've considered

Setting the AC Power via the Anker app. But I can't see that reflected in some API request. I must add that I have limited knowledge of the API.

### Additional context

None

Jorre05, 2025-09-10:
Thanks for the clarification.

Jorre05, 2025-09-17:
How nice!

thomluther, 2026-01-29:
Those controls are hybrid controls that required Api commands to change the station panels in the Api and MQTT commands to change the settings for the device as well.
This is implemented with version 3.5.0