thomluther/ha-anker-solix #360: Number entity Solarbank 2 "Einspeisevorgabe" / feed-in target is gone since version 3.x

markusmauch, 2025-09-27:
### System Health details

## System Information

version | core-2025.9.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.12.43-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2263
Downloaded Repositories | 25

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.09.0
agent_version | 1.7.2
docker_version | 28.3.3
disk_total | 30.8 GB
disk_used | 16.8 GB
nameservers | 192.168.0.4
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Cloudflared (5.3.9), Zigbee2MQTT (2.6.1-1), MQTT Explorer (browser-1.0.1), Studio Code Server (5.19.3), Music Assistant Server (2.6.0), Terminal & SSH (9.20.1), Samba share (12.5.2), Mosquitto broker (6.5.2), AppDaemon (0.17.9), keyble-mqtt (1.5.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 8
views | 39
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethce6c72a (disabled), vethb2eadf5 (disabled), veth080dbc2 (disabled), veth19f067d (disabled), vethd611d49 (disabled), veth774cedf (disabled), vethae75928 (disabled), veth081b1fe (disabled), veth84297a4 (disabled), vethf1afd9d (disabled), veth54067ed (disabled), vethab16ca5 (disabled), vethc2a5bea (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 (192.168.0.8/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethce6c72a (), vethb2eadf5 (), veth080dbc2 (), veth19f067d (), vethd611d49 (), veth774cedf (), vethae75928 (), veth081b1fe (), veth84297a4 (), vethf1afd9d (), veth54067ed (), vethab16ca5 (), vethc2a5bea ()
ipv6_addresses | lo (::1/128), enp0s18 (fe80::b2:ebaa:a44f:150d/64), hassio (fe80::8d6:7ff:fe6f:15d2/64), docker0 (fe80::9017:2cff:fe04:931d/64), vethce6c72a (fe80::8cd9:c6ff:fe1e:2f85/64), vethb2eadf5 (fe80::40f9:17ff:fe82:8ae8/64), veth080dbc2 (fe80::b492:43ff:fed6:aa6e/64), veth19f067d (fe80::a0c3:1cff:fe86:24b1/64), vethd611d49 (fe80::f86b:6bff:fe3d:244b/64), veth774cedf (fe80::b4f8:4aff:fe9e:5b76/64), vethae75928 (fe80::10ee:1cff:fe69:c973/64), veth081b1fe (fe80::3cfc:fdff:fe76:df92/64), veth84297a4 (fe80::383d:c3ff:fec8:68dc/64), vethf1afd9d (fe80::498:73ff:feea:cc5d/64), veth54067ed (fe80::9463:80ff:fe8e:6e2b/64), vethab16ca5 (fe80::4f3:feff:fe4c:31a5/64), vethc2a5bea (fe80::a8a9:61ff:fe2c:c075/64)
announce_addresses | 192.168.0.8, fe80::b2:ebaa:a44f:150d

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 13. September 2025 um 15:43
-- | --
current_recorder_run | 25. September 2025 um 21:39
estimated_db_size | 1322.98 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Integration version

3.x

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I successfully managed zero feed-in mode by reading my smart meter and setting the feed-in target of the Solarbank 2 Pro by a Home Assistant automation. Now the number entity (number.solarbank_2_e1600_pro_system_einspeisevorgabe) is gone. Is there some replacement? Im aware of the limitations by doing this not the official way, but it worked surprisingly well.

### Reproduction steps

1.Update to 3.x of the integration
2. Run a template in the DevTools: {{states.number.solarbank_2_e1600_pro_system_einspeisevorgabe.state}}

### Debug logs

```text
-
```

### Diagnostics dump

-

AchimTheobald, 2025-09-27:
Search for entities with "preset" in name.

My system runs in denglish modus. Some entities in german the other in english.

[image]

[image]

thomluther, 2025-09-27:
> Forgot to mention that I have a combined SB1 + SB2Pro setup, though I hope not that this makes the difference...

That can surely make the difference. I think it will be easiest if you provide an export
[Export systems action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action)

> Where would I find the logs for the integration? Seems to be nothing in Settings, System, Protocols, Home Assistant Core...

As for each integration, you have to enable debug mode in the integration. Once you disable it again, you will automatically get the logs since enabling debug.

markusmauch, 2025-09-28:
Sure, I have a combined system with the Solarbank 2 E1600 Pro as master and the Solarbank E1600 as slave. No additional batteries or any other gear. I'm using a Home Assistant automation that uses the values of a SML IR reader to control the amount of energy the system feeds into my home.

markusmauch, 2025-10-01:
I can confirm that it works correctly now with version 3.3.2. Thanks for you great work, @thomluther. If you need any assistance, let me know...

N1co93, 2025-10-01:
> Actually I don't know what should be missing. I can see all entities when loading the exported configuration.
> 
> > on the system of my father the data is still missing with Version 3.3.2.
> 
> So what data is supposed to be missing?

If I understand correctly, this entity shouldn't show 'unknown' but rather the maximum discharge power (in Watts) the battery is allowed to supply to the house.

Or is my understanding incorrect?

[image]

thomluther, 2025-10-01:
No, this is the output preset for custom Mode. It has no value in automode on purpose, since changing it would not have any effect but only unnecessarily Api calls